The-OpenROAD-Project/OpenROAD-flow-scripts #2: Resizer Segfaults if setRC.tcl is not present

quentinw91, 2020-10-20:
Hi,
Nice to see that the OpenRoad-flow repo has been reopened !
Small bug with ```resize.tcl``` script: If the setRC.tcl is removed from the platform directory, the ```repair_design``` command dumps a segmentation fault.

Can be reproduced with the nangate45 gcd example, with OpenRoad-flow-scripts version 85bc23d5e.

```
bash-4.2$ make place
[...]
Notice 0: Finished DEF file: ./results/nangate45/gcd/3_2_place_iop.def
Warning: wire capacitance is zero. Use the set_wire_rc command to set wire resistance and capacitance.

==========================================================================
report_checks
--------------------------------------------------------------------------
Startpoint: _512_ (rising edge-triggered flip-flop clocked by core_clock)
Endpoint: _503_ (rising edge-triggered flip-flop clocked by core_clock)
Path Group: core_clock
Path Type: max

  Delay    Time   Description
---------------------------------------------------------
   0.00    0.00   clock core_clock (rise edge)
   0.00    0.00   clock network delay (ideal)
   0.00    0.00 ^ _512_/CK (DFF_X1)
   0.08    0.08 ^ _512_/Q (DFF_X1)
   0.04    0.13 ^ _305_/Z (BUF_X1)
   0.07    0.20 ^ _308_/Z (XOR2_X1)
   0.02    0.22 v _311_/ZN (AOI21_X1)
   0.05    0.27 ^ _312_/ZN (OAI21_X1)
   0.03    0.30 v _315_/ZN (AOI21_X1)
   0.05    0.34 ^ _316_/ZN (OAI21_X1)
   0.03    0.37 v _319_/ZN (AOI21_X1)
   0.04    0.42 ^ _320_/ZN (OAI21_X1)
   0.03    0.44 v _322_/ZN (OAI21_X1)
   0.05    0.49 ^ _324_/ZN (OAI21_X1)
   0.03    0.52 v _325_/ZN (OAI21_X1)
   0.06    0.57 ^ _327_/ZN (OAI21_X1)
   0.04    0.62 v _328_/ZN (OAI221_X1)
   0.05    0.67 v _329_/ZN (AND3_X1)
   0.06    0.72 ^ _330_/ZN (OAI22_X1)
   0.03    0.75 v _331_/ZN (AOI21_X1)
   0.05    0.80 ^ _332_/ZN (OAI21_X1)
   0.10    0.90 ^ _338_/ZN (AND4_X1)
   0.08    0.98 ^ _354_/Z (BUF_X1)
   0.02    1.00 v _383_/ZN (AOI22_X1)
   0.04    1.03 ^ _386_/ZN (OAI22_X1)
   0.00    1.03 ^ _503_/D (DFF_X1)
           1.03   data arrival time

   2.00    2.00   clock core_clock (rise edge)
   0.00    2.00   clock network delay (ideal)
   0.00    2.00   clock reconvergence pessimism
           2.00 ^ _503_/CK (DFF_X1)
  -0.04    1.96   library setup time
           1.96   data required time
---------------------------------------------------------
           1.96   data required time
          -1.03   data arrival time
---------------------------------------------------------
           0.93   slack (MET)

==========================================================================
report_tns
--------------------------------------------------------------------------
tns 0.00

==========================================================================
report_wns
--------------------------------------------------------------------------
wns 0.00

==========================================================================
report_design_area
--------------------------------------------------------------------------
Design area 438 u^2 7% utilization.

==========================================================================
instance_count
--------------------------------------------------------------------------
386

==========================================================================
pin_count
--------------------------------------------------------------------------
964

Perform port buffering...
Perform buffer insertion...
0:00.84elapsed 83%CPU 66268memKB
make: *** [results/nangate45/gcd/3_3_place_resized.def] Error 11
```

hongzhi2015, 2020-10-22:
I also encounter a crash at `resize.tcl: repair_design`.
At my site, gcd goes through the flow, ibex example always crash.

How to reproduce

Under OpenROAD-flow-scripts/flow:

```bash
$ docker run -it --rm --network=host \
  --mount type=bind,src=`pwd`/designs,dst=/OpenROAD-flow/flow/designs \
  --mount type=bind,src=`pwd`/platforms,dst=/OpenROAD-flow/flow/platforms \
  --user `id -u`:`id -g` \
  --workdir /OpenROAD-flow/flow \
  --entrypoint /bin/bash \
  openroad/flow:20201021 \
  -c "source /OpenROAD-flow/setup_env.sh && make DESIGN_CONFIG=./designs/sky130/ibex/config.mk V=1"
```
 
...
Perform port buffering...
Perform buffer insertion...
0:06.55elapsed 49%CPU 205784memKB
make: *** [results/sky130/ibex/3_3_place_resized.def] Error 11

Directory designs and platforms must be mounted, because sky130 directories are not copyed to docker image at build time.
openroad/flow:20201021 (tagged with date as image version) is the latest image build by OpenROAD-flow-scripts/build_openroad.sh

Crash location script/resize.tcl:
```tcl
puts "Perform buffer insertion..."
set_max_fanout $::env(MAX_FANOUT) [current_design]
repair_design -max_wire_length $::env(MAX_WIRE_LENGTH) -buffer_cell $buffer_cell
```

By modifing the Makefile and running the tcl manually, a segfault is revealed:
Perform port buffering...
Perform buffer insertion...
make: *** [results/sky130/ibex/3_3_place_resized.def] Segmentation fault (core dumped)

jjcherry56, 2020-10-29:
fixed in 08365fcd resizer missing set_wire_rc seg fault
hongzhi2015: the sky130 platform is broken because it is missing setRC.tcl.
You can add the following to avoid the bug until it gets pushed to the public repo:
set_wire_rc -layer met2

closed 2020-10-29 by jjcherry56