The-OpenROAD-Project/OpenROAD-flow-scripts #4: nangate45/tinyRocket does not place SRAMs

davidmetz, 2020-11-06:
**Describe the bug**
 [Design] [Flow]
When running generating the tinyRocket sample design (`make DESIGN_CONFIG=./designs/nangate45/tinyRocket/config.mk`) SRAMs are not placed.
[2_2_floorplan_io.log](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/5501545/2_2_floorplan_io.log)
I also checked the final GDS and it does not include the dummy SRAMs.
I encountered this because I am trying to use OpenRAM generated SRAMs fro Rocket.
After realizing that they were not included I checked the example design and observed the same behavior.
**Expected behavior**
SRAM macros should be placed

**Environment (please complete the following information):**
docker under linux mint - generated with current master (872b051)

rovinski, 2020-11-09:
The placement errors are actually false errors. Macros are placed in step 2_4. You should be able to open up any DEF after 2_4 and see the macros placed.

FakeRAM does not have GDS views and is a pitfall of the nangate45 process. You will also see the following error during 6_1 GDS merge:
```
[ERROR] LEF Cell 'fakeram45_64x32' has no matching GDS cell. Cell will be empty
```
LEF & Lib views exist, however, and it is useful to use them up until GDS.

If you are using OpenRAM in order to get back-end views, you probably need to add the GDS files to your config file.

```
export ADDITIONAL_GDS  = PATH/TO/RAM1.gds \
                         PATH/TO/RAM2.gds
```

rovinski, 2020-11-13:
Is this still an issue? Or can it be closed?

davidmetz, 2020-11-19:
Hi - thanks for the reply! You are right - they are placed.

closed 2020-11-19 by davidmetz