The-OpenROAD-Project/OpenROAD-flow-scripts #14: OpenRoad docker build error

raghu1996, 2021-04-04:
Hi,
I am trying to do the docker build by executing ./build_openroad.sh.
However the build errors out with the following message:

g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [sec/OpenDB/src/lefin/CMakeFiles/lefin.dir/build.make:183: src/OpenDB/src/lefin/CMakeFiles/lefin.dir/lefTechLayerCutSpacingParser.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....

How to resolve this?

vvbandeira, 2021-04-05:
Hi, I cannot reproduce the error on master branch.
Can you paste the output of the following commands?
```
uname -a
docker --version
git status
git show
```

raghu1996, 2021-04-05:
Hi,
Outputs:

 uname -a
Linux localhost.localdomain 3.10.0-1160.6.1.el7.x86_64 #1 SMP Tue Nov 17 13:59:11 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

 docker --version
Docker version 20.10.5, build 55c4c88

 git status
# On branch master
nothing to commit, working directory clean

 git show
commit 59ad47a1325239b578bf1c2b3cf6617e44d05d47
Merge: c405b57 fe99622
Author: Vitor Bandeira <vvbandeira@users.noreply.github.com>
Date:   Fri Apr 2 12:00:59 2021 -0300

    Merge pull request #112 from The-OpenROAD-Project-private/adding-metrics-checks
    
    Adding metrics checks

vvbandeira, 2021-04-05:
On my test machine (also a Centos7) I am using the same docker version and commit sha without issue. The only difference being the kernel version, which should not cause this error.
```
Linux machine_name 3.10.0-1160.15.2.el7.x86_64 #1 SMP Wed Feb 3 15:06:38 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
```

Could you run the following commands and upload the `output.txt` file generated into https://pastebin.pl/ or a similar site?

```
git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts flow-issue-14
cd flow-issue-14
./build_openroad.sh &> output.txt
```

raghu1996, 2021-04-07:
Hi
Output log had error message saying virtual memory is out of space. I am trying to build after increasing the memory. Will keep you posted!

raghu1996, 2021-04-08:
The issue is resolved after increasing the virtual memory. Thanks!

closed 2021-04-08 by raghu1996