The-OpenROAD-Project/OpenROAD-flow-scripts #17: Error while loading shared libraries: libtclreadline-2.1.0.so

Raynchowkw, 2021-04-20:
I "git pull" openroad-flow-scripts and build it again.
When I "cd flow; make", I got error message in terminal outputs.
It is in "17. Executing Verilog backend." It says
"
openroad: error while loading shared libraries: libtclreadline-2.1.0.so: cannot open shared object file: No such file or directory
make: *** [results/nangate45/gcd/2_1_floorplan.def] Error 127
"

maliberty, 2021-04-20:
I think you need to set your LD_LIBRARY_PATH to wherever it is installed.

stevvik, 2021-04-28:
Hello! I have the same error running flow in docker container. Docker image was created successfully, but flow fails with the error described above. As I understand the installation of this lib is missed in Dockerfile.

maliberty, 2021-04-28:
Are you seeing it as missing when you are inside the docker container or outside?  It will only be visible inside the container.  It can't be both linked successfully and missing.

stevvik, 2021-04-28:
Of course it is missimg inside container! The "make" inside flow directory in container returns this error. And it helps to copy this libtclreadline-2.1.0.so in /usr/lib64/ directory inside container and I even checked in this change in image! But it would be better to add it's installation in Dokerfile.

Raynchowkw, 2021-04-28:
> Of course it is missimg inside container! The "make" inside flow directory in container returns this error. And it helps to copy this libtclreadline-2.1.0.so in /usr/lib64/ directory inside [...]

Hi, How to copy libtclreadline-2.1.0.so in /usr/lib64/ inside container?

maliberty, 2021-04-28:
The problem is that there are multiple containers involved.  When openroad builds it is in a container with tclreadline installed.  The build result is then copied to a new container which doesn't have it installed.

Vitor will fix the issue but in the short term you can work around it with:
   docker run -it openroad/flow
    yum install -y https://dl.fedoraproject.org/pub/epel/epel-release-latest-7.noarch.rpm
    yum install -y libgomp python36-libs qt5-qtbase-devel tcl-tclreadline
    yum update -y
   ./tools/build/OpenROAD/src/openroad

closed 2021-04-28 by maliberty

stevvik, 2021-04-28:
Maliberty, please, can you explain why build results are copied to a new container?

maliberty, 2021-04-28:
Each of the individual tools (yosys & openroad) has its own docker file so it can be built standalone.  The flow just invokes them to build the tools and copies the build result out.  It avoids repeating the whole Dockerfile again in the flow's Dockerfile which would inevitably lead to synchronization issues.

stevvik, 2021-04-28:
Now I see, thank you, Maliberty.

Raynchowkw, 2021-04-30:
> The problem is that there are multiple containers involved. When openroad builds it is in a container with tclreadline installed. The build result is then copied to a new container which doesn't [...]

I am not familiar with docker. I notice that `docker run -it openroad/flow` is different from the corresponding line in README that is `docker run -it -u $(id -u ${USER}):$(id -g ${USER}) -v $(pwd)/flow/platforms:/OpenROAD-flow/flow/platforms:ro openroad/flow bash`.
Which one I should use?
Additionally, I find that after I ran `docker run -it` once and use `ctrl d` to exit, I ran it again. I found the left-handside initiator in terminal is different (root@xxxx_different number_). Does it mean I create a new image or container? Do I need to install tclreadline each time after I run `docker run -it openroad/flow`? It costs a considerable time.
Thanks.

vvbandeira, 2021-04-30:
Are you seeing the need to install the package after the commit 7c5cd0f ?

For your reference: https://stackoverflow.com/a/26960888

Also, you could create a persistent container take a look into the following commands:
```
# to start a container
docker start --help
# to "get inside the container"
docker attach --help
# to "detach / exit temporarily", inside the container press
ctrl-p ctrl-q
```

Raynchowkw, 2021-04-30:
> Are you seeing the need to install the package after the commit [7c5cd0f](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/commit/7c5cd0fc05fcea0c17a72f5ea09c065e1fcfbcde) ?

I noticed your new commit. I think the added step of "./DependencyInstaller.sh" is exactly what deals with my original problem of missing libtclreadline-2.1.0.so.

---

I find that if I just run `docker run -it openroad/flow`, /flow is not completed and I cannot run `make`.
[image]
After I follow back the README one: `docker run -it -u $(id -u ${USER}):$(id -g ${USER}) -v $(pwd)/flow/platforms:/OpenROAD-flow/flow/platforms:ro openroad/flow bash`
it gets correct and I can run `make`.
[image]

---
However, I find tools module is not completed. I mean, some src files are missing compared to github. For example, I want to find TritonRoute src to make some changes, but could not. It happens in both cases of `docker run -it openroad/flow` and `docker run -it -u $(id -u ${USER}):$(id -g ${USER}) -v $(pwd)/flow/platforms:/OpenROAD-flow/flow/platforms:ro openroad/flow bash`
[image]
[image]

---
Just to provide more information in case it's needed.
At first, I just did `git pull` under my downloaded directory of "openroad-flow-scripts" first, **bypassing the step of `./build_openroad.sh`** because I built it before, and then directly applied Malibery's [solutions](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/17#issuecomment-828606486). Just after I found the problem I described above, I built it again, but the problem persists. 
I am not sure if I need to build it again after I do `git pull`. Maybe this time I messed it up.
I want to find the src in TritonRoute and make some changes to it, and then run the flow to output some extra information from it.
Thanks.