The-OpenROAD-Project/OpenROAD-flow-scripts #23: Environment variable ABC_CLOCK_PERIOD_IN_PS in synthesis script

animeshbchowdhury, 2021-05-13:
Setting environment variable ABC_CLOCK_PERIOD_IN_PS

I want to generate variant of synthesized files only (no placement and routing). I use `make synth` to generate the synthesized netlist and change 'clock period' in constraint.sdc file to generate a lot of variants. However, varying the clock period is generating the same synthesized netlist.

After looking into the synth.tcl script, I deduced that the variable `ABC_CLOCK_PERIOD_IN_PS` is responsible to synthesize netlist satisfying the timing constraint. However, I failed to find the relevant code where this environmental variable is set.

Can you help me how constraint.sdc file information may be used to set this variable?

maliberty, 2021-05-13:
It is optional but you can see an example at https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/2ab12af6646556fc72bd18e0feb47ac4d61a2aed/flow/platforms/nangate45/config.mk#L40

animeshbchowdhury, 2021-05-13:
Thanks, this helps a lot!

rovinski, 2021-05-25:
Is there anything else to be done here? Or can this issue be closed?

animeshbchowdhury, 2021-05-25:
This issue can be closed. Thanks for the help.

Best Regards,
Animesh

closed 2021-05-25 by rovinski