The-OpenROAD-Project/OpenROAD-flow-scripts #25: Failed to run command 'klayout': No such file or directory

Raynchowkw, 2021-05-18:
After running `make`, it meets Error when it run `klayout -zz` I guess.
```
(/usr/bin/time -f "%Eelapsed %PCPU %MmemKB" stdbuf -o L klayout -zz -rd design_name=gcd \
        -rd in_def=results/nangate45/gcd/base/6_final.def \
        -rd in_files="./platforms/nangate45/gds/NangateOpenCellLibrary.gds  " \
        -rd config_file= \
        -rd seal_file="" \
        -rd out_file=results/nangate45/gcd/base/6_1_merged.gds \
        -rd tech_file=./objects/nangate45/gcd/base/klayout.lyt \
        -rm ./util/def2stream.py) 2>&1 | tee ./logs/nangate45/gcd/base/6_1_merge.log
stdbuf: failed to run command ‘klayout’: No such file or directory
0:00.01elapsed 6%CPU 784memKB
make: *** [Makefile:523: results/nangate45/gcd/base/6_1_merged.gds] Error 127
```
How to check: Did I install klayout well? Did I set up environment for klayout correctly?
Thanks.

maliberty, 2021-05-18:
It doesn't seem like it.  You can install it with

yum install -y https://www.klayout.org/downloads/CentOS_7/klayout-0.26.4-0.x86_64.rpm

closed 2021-05-18 by maliberty

Raynchowkw, 2021-05-18:
After I install it, how to make it searchable by 'make' in the flow? Do I need to build the flow again by `./build_openroad.sh --local` so that I can integrate it into the flow?

maliberty, 2021-05-18:
you just need it in your PATH

Raynchowkw, 2021-05-18:
Thanks. Got it!
I just installed it and didn't need to add it to PATH manually because it's already in /bin.
So I can run through the flow now.