The-OpenROAD-Project/OpenROAD-flow-scripts #49: locally build issue

Raynchowkw, 2021-06-01:
I have successfully built openroad-flow-scripts before. This time, I forked it to my own repo and made some modifications. And I used `git clone --recursive` to download it to another directory than the previous successfully built one. 
When I run `./build_openroad.sh --local`, it just reach [89%] and get an error.
```
[ 89%] Built target PartitionMgr
make[2]: Leaving directory '/home/rayncc/my_opr/OpenROAD-flow-scripts/tools/build/OpenROAD'
[ 89%] Built target psm
make[1]: Leaving directory '/home/rayncc/my_opr/OpenROAD-flow-scripts/tools/build/OpenROAD'
make: *** [Makefile:133: all] Error 2
```
Could anyone help me? Thanks.

maliberty, 2021-06-01:
you have to actually include the error ....

Raynchowkw, 2021-06-01:
I did `git checkout master` in my tools/Openroad and ran it again. Now it stopped at 
```
[ 93%] Linking CXX static library TritonCTS.a
cd /home/rayncc/my_opr/OpenROAD-flow-scripts/tools/build/OpenROAD/src/TritonCTS/src && /usr/local/bin/cmake -P CMakeFiles/TritonCTS.dir/cmake_clean_target.cmake
cd /home/rayncc/my_opr/OpenROAD-flow-scripts/tools/build/OpenROAD/src/TritonCTS/src && /usr/local/bin/cmake -E cmake_link_script CMakeFiles/TritonCTS.dir/link.txt --verbose=1
/opt/rh/devtoolset-8/root/usr/bin/ar qc TritonCTS.a  CMakeFiles/TritonCTS.dir/CMakeFiles/TritonCTS.dir/tritonctsTCL_wrap.cxx.o CMakeFiles/TritonCTS.dir/TritonCTS-tclInitVar.cc.o CMakeFiles/TritonCTS.dir/Clock.cpp.o CMakeFiles/TritonCTS.dir/HTreeBuilder.cpp.o CMakeFiles/TritonCTS.dir/MakeTritoncts.cpp.o CMakeFiles/TritonCTS.dir/PostCtsOpt.cpp.o CMakeFiles/TritonCTS.dir/SinkClustering.cpp.o CMakeFiles/TritonCTS.dir/StaEngine.cpp.o CMakeFiles/TritonCTS.dir/TechChar.cpp.o CMakeFiles/TritonCTS.dir/TritonCTS.cpp.o CMakeFiles/TritonCTS.dir/clustering.cpp.o CMakeFiles/TritonCTS.dir/LevelBalancer.cpp.o
/opt/rh/devtoolset-8/root/usr/bin/ranlib TritonCTS.a
make[2]: Leaving directory '/home/rayncc/my_opr/OpenROAD-flow-scripts/tools/build/OpenROAD'
[ 93%] Built target TritonCTS
make[1]: Leaving directory '/home/rayncc/my_opr/OpenROAD-flow-scripts/tools/build/OpenROAD'
make: *** [Makefile:133: all] Error 2
```

maliberty, 2021-06-01:
again no error message....

Raynchowkw, 2021-06-01:
OK. I find those by searching in terminal. I think it is because my modifications of TritonRoute src code.
```
/home/rayncc/my_opr/OpenROAD-flow-scripts/tools/OpenROAD/src/TritonRoute/src/dr/FlexDR.cpp: In member function ‘int fr::FlexDR::main()’:
/home/rayncc/my_opr/OpenROAD-flow-scripts/tools/OpenROAD/src/TritonRoute/src/dr/FlexDR.cpp:1996:35: error: request for member ‘get’ in ‘instTerm’, which is of pointer type ‘fr::frInstTerm* const’ (maybe you meant to use ‘->’ ?)
       if (isSkipInstTerm(instTerm.get())) {
                                   ^~~
/home/rayncc/my_opr/OpenROAD-flow-scripts/tools/OpenROAD/src/TritonRoute/src/dr/FlexDR.cpp:1996:11: error: ‘isSkipInstTerm’ was not declared in this scope
       if (isSkipInstTerm(instTerm.get())) {
           ^~~~~~~~~~~~~~
/home/rayncc/my_opr/OpenROAD-flow-scripts/tools/OpenROAD/src/TritonRoute/src/dr/FlexDR.cpp:1996:11: note: suggested alternative: ‘instTerm’
       if (isSkipInstTerm(instTerm.get())) {
           ^~~~~~~~~~~~~~
           instTerm
/home/rayncc/my_opr/OpenROAD-flow-scripts/tools/OpenROAD/src/TritonRoute/src/dr/FlexDR.cpp:2005:8: error: ‘inst’ was not declared in this scope
        inst->getTransform(shiftXform);
        ^~~~
/home/rayncc/my_opr/OpenROAD-flow-scripts/tools/OpenROAD/src/TritonRoute/src/dr/FlexDR.cpp:2005:8: note: suggested alternative: ‘init’
        inst->getTransform(shiftXform);
        ^~~~
        init
```

maliberty, 2021-06-01:
There is no such code in the current master branch.  I don't know what you are working on but the problem seems to be local.

closed 2021-06-01 by maliberty

Raynchowkw, 2021-06-01:
Yes I think so. Sorry about that. I made some modification of the code. And I show that in the discussion [#625](https://github.com/The-OpenROAD-Project/OpenROAD/discussions/625). Let's discuss this there. Thanks.