The-OpenROAD-Project/OpenROAD-flow-scripts #54: Can't find object for defparam `EnableDataIntgGen`!

peijunh, 2021-06-13:
Hello! When I run "make synth" in the OpenROAD-flow-scripts/flow , I have the following error reported:

75.2.59. Executing AST frontend in derive mode using pre-parsed AST for module `\tlul_adapter_reg'.
/home/houpeijun/peijunh/opentitan_sverilog/prim_subreg.v:0: ERROR: Can't find object for defparam `EnableDataIntgGen`!
make: *** [results/sky130hd/opentitan_soc/base/1_1_yosys.v] Error 1

While the parameter EnableDataIntgGen is neither used in module tlul_adapter_reg nor in prim_subreg.

So could you share some advice on what might be the possible  reason for this error? Thanks!

maliberty, 2021-06-13:
Can you provide a test case?

maliberty, 2021-06-13:
You may also want to ask https://github.com/YosysHQ/yosys

msaligane, 2021-06-13:
Yes, I think Peijun was supposed to file an issue there.

closed 2021-06-13 by msaligane