The-OpenROAD-Project/OpenROAD-flow-scripts #64: [Build] Issue while building the latest version of OpenROAD-flow-scripts

naomimotwani98, 2021-06-26:
Build Issue 

I encountered the following issue while building the latest version of OpenROAD-flow-scripts with LS Oracle integration. 

Scanning dependencies of target lsoracle
[ 96%] Building CXX object core/CMakeFiles/lsoracle.dir/lsoracle.cpp.o
[100%] Linking CXX executable lsoracle
CMakeFiles/lsoracle.dir/lsoracle.cpp.o: In function `int alice::show_command<std::shared_ptr<mockturtle::names_view<mockturtle::aig_network> >, std::shared_ptr<oracle::partition_manager<mockturtle::names_view<mockturtle::aig_network> > >, std::shared_ptr<mockturtle::names_view<mockturtle::mig_network> >, std::shared_ptr<oracle::partition_manager<mockturtle::names_view<mockturtle::mig_network> > >, std::shared_ptr<mockturtle::names_view<mockturtle::xag_network> >, std::shared_ptr<mockturtle::names_view<mockturtle::klut_network> > >::show_store<std::shared_ptr<mockturtle::names_view<mockturtle::klut_network> > >()':
lsoracle.cpp:(.text._ZN5alice12show_commandIJSt10shared_ptrIN10mockturtle10names_viewINS2_11aig_networkEEEES1_IN6oracle17partition_managerIS5_EEES1_INS3_INS2_11mig_networkEEEES1_INS8_ISC_EEES1_INS3_INS2_11xag_networkEEEES1_INS3_INS2_12klut_networkEEEEEE10show_storeISL_EEiv[_ZN5alice12show_commandIJSt10shared_ptrIN10mockturtle10names_viewINS2_11aig_networkEEEES1_IN6oracle17partition_managerIS5_EEES1_INS3_INS2_11mig_networkEEEES1_INS8_ISC_EEES1_INS3_INS2_11xag_networkEEEES1_INS3_INS2_12klut_networkEEEEEE10show_storeISL_EEiv]+0x2ab): warning: the use of `tmpnam' is dangerous, better use `mkstemp'
../lib/mockturtle/lib/abcsat/liblibabcsat.a(satSolver.cpp.o): In function `pabc::sat_solver_reducedb(pabc::sat_solver_t*)':
satSolver.cpp:(.text+0x3f8c): undefined reference to `clock_gettime'
satSolver.cpp:(.text+0x46e7): undefined reference to `clock_gettime'
../lib/mockturtle/lib/abcsat/liblibabcsat.a(satSolver.cpp.o): In function `pabc::sat_solver_search(pabc::sat_solver_t*, long)':
satSolver.cpp:(.text+0x68eb): undefined reference to `clock_gettime'
../lib/mockturtle/lib/abcsat/liblibabcsat.a(satSolver.cpp.o): In function `pabc::sat_solver_solve_internal(pabc::sat_solver_t*)':
satSolver.cpp:(.text+0x8841): undefined reference to `clock_gettime'
satSolver.cpp:(.text+0x88bb): undefined reference to `clock_gettime'
collect2: error: ld returned 1 exit status
gmake[2]: *** [core/CMakeFiles/lsoracle.dir/build.make:89: core/lsoracle] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1681: core/CMakeFiles/lsoracle.dir/all] Error 2

Are there any dependencies that need to be built before? Or something I am missing?

maliberty, 2021-06-26:
what OS & compiler are you using?

naomimotwani98, 2021-06-27:
This is what I am currently using 

Description:	Red Hat Enterprise Linux Server release 6.10 (Santiago)
Release:	6.10
gcc version 4.4.7

maliberty, 2021-06-28:
I don't have access to the platform but from the message it seems like you need to link with librt.  Can you try that?

maliberty, 2021-06-28:
We do have in our cmake

if (CMAKE_CXX_COMPILER_ID STREQUAL "GNU"
    AND CMAKE_SYSTEM_NAME STREQUAL "Linux")
  find_library(LIBRT rt)
  target_link_libraries(openroad ${LIBRT})
endif()

Perhaps this needs to trigger on your platform too.

naomimotwani98, 2021-06-30:
Hi! Sorry for the late response, I didn't get where this should be located.
do you mean the cmake in OpenROAD-flow-scripts/tools/OpenROAD?

rovinski, 2021-06-30:
Yes, that seems right. If you get it to work, submit a pull request or open an issue in the [LSOracle source repository](https://github.com/lnis-uofu/LSOracle). We will eventually upstream the changes here.

maliberty, 2021-06-30:
I am guessing but in LSOracle/CMakeLists.txt you need to add the above with openroad changed to lsoracle

naomimotwani98, 2021-07-03:
Hi! Unfortunately that did not work - I added the following in LSOracle/CMakeLists.txt -

if (CMAKE_CXX_COMPILER_ID STREQUAL "GNU"
AND CMAKE_SYSTEM_NAME STREQUAL "Linux")
find_library(LIBRT rt)
target_link_libraries(lsoracle ${LIBRT})
endif()

I still got the same error 
is there any other workaround?

maliberty, 2021-07-03:
do you see -lrt being added to the link invocation?

naomimotwani98, 2021-07-03:
Hey adding the following lines to tools/LSOracle/core/CMakeLists.txt seems to work, It did not throw any error 

if (CMAKE_CXX_COMPILER_ID STREQUAL "GNU"
AND CMAKE_SYSTEM_NAME STREQUAL "Linux")
find_library(LIBRT rt)
target_link_libraries(lsoracle ${LIBRT})
endif()

closed 2021-07-03 by naomimotwani98