The-OpenROAD-Project/OpenROAD-flow-scripts #66: [Flow] LSoracle integration appears incomplete

arlpetergadfort, 2021-06-29:
**Describe the bug**
When using the LSORACLE, there are 2 issues.
1. lsoarcle in yosys is excepting some arguments, but the command inserted into the synth.tcl script does not include them so the it generates this message:
```
5. Executing LSOracle pass (MIG optimization using LSOracle).
checking arguments
Invalid number of arguments
```
Which does not describe the missing / required arguments what are invalid.

2. When arguments are added to the lsoracle fails with the following message (this issue may belong in the LSoracle repo, but adding it here for now):
```
ERROR: LSOracle: execution of command "lsoracle -f /tmp/yosys-abc-c3g61p/lso.script 2>&1" failed: return code 255
```

Manually running that command seems to fail on:
```
Could not load context file at: /usr/local/share/lsoracle/test.ini
```
This seems to stem from an assumption that the test.ini file has been installed in a very particular location. This appears to be hardcoded in lsoracle and probably needs to be corrected.

**Expected behavior**
After running `make synth USE_LSORACLE=1` I would except lsoracle to have be part of the synthesis.

**Environment (please complete the following information):**
 - OS: REHL7 and Ubuntu 20
 - OpenROAD-flow commit: 61afe66

maliberty, 2021-06-29:
@yashton would you take a look at this?

yashton, 2021-06-29:
I'll look into it first thing.

ibrahimkhairy, 2021-06-30:
I have the same issue at my end. 

It seems the installed LSOracle source files have a dependency over a hardcoded tool config file;  this hardcoded dependency is located at the following script:
https://github.com/The-OpenROAD-Project-private/LSOracle/blob/master/core/commands/optimization/oracle.hpp

at the following line: 
`      std::string config_file{"/usr/local/share/lsoracle/test.ini"}; `

PR #67 (2021-07-02): Add explicit script and configuration variables for LSOracle in synth script
Fixes #66 
Added additional variables to the makefile to specify the lsoracle binary and config file locations, defaulting to relative on the filesystem. Additionally I made the lsoracle recipe explicit in the synth script, like how abc is invoked.
closed 2021-07-02 by rovinski