The-OpenROAD-Project/OpenROAD-flow-scripts #78: Difference between "CELL_PAD_IN_SITES_GLOBAL_PLACEMENT", "CELL_PAD_IN_SITES_DETAIL_PLACEMENT"

peijunh, 2021-07-12:
Hello, I am writing to ask about the difference between this two commands:  "CELL_PAD_IN_SITES_GLOBAL_PLACEMENT", and "CELL_PAD_IN_SITES_DETAIL_PLACEMENT"

This two commands are from ./OpenROAD-flow-scripts/flow/designs/$(platform)/$(design)/config.mk

I know they are to keep distance between standard cell but what is their exact difference? Like, are they just happen in different stages in APR?

I appreciate any help and advice, thanks!

maliberty, 2021-07-12:
They are for different stages as the names imply: global placement and detailed placement.   The general idea is to allow the detailed placer to use a smaller value gives it more room to work with.  In principle they can both be equal though that will make detailed placement more challenging.

peijunh, 2021-07-13:
Thanks a lot, I will use CELL_PAD_IN_SITES_GLOBAL_PLACEMENT then!

closed 2021-07-13 by peijunh