The-OpenROAD-Project/OpenROAD-flow-scripts #89: [Flow] PDN error building coyote_tc sky130hd example

nmoroze, 2021-07-20:
**Describe the bug**

When building the coyote_tc example for sky130hd (by uncommenting `DESIGN_CONFIG=./designs/sky130hd/coyote_tc/config.mk` in the Makefile and running `make`), I get the following error during PDN generation:

```
[WARNING PDN-0099] No power/ground pads found on bottom edge
[WARNING PDN-0099] No power/ground pads found on right edge
[WARNING PDN-0099] No power/ground pads found on top edge
[WARNING PDN-0099] No power/ground pads found on left edge
[ERROR PDN-0099] Cannot place core rings without pwr/gnd pads on each side
```

**Expected behavior**

I'd expect this example design to build fully. It seems like power and ground pads are part of the design.

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - OpenROAD-flow commit f85c8a94ff620dd459a5d0255d5ac040fc4102cf

Thanks for taking a look!

tspyrou, 2021-10-01:
@ibrahimkhairy since this is sky130 can you take a look at it while Colin is busy on other issues?

ibrahimkhairy, 2021-10-01:
> @ibrahimkhairy since this is sky130 can you take a look at it while Colin is busy on other issues?

Yes, sure. I'll have a look.

tspyrou, 2021-11-02:
@ibrahimkhairy did you find anything?

vijayank88, 2022-01-04:
@tspyrou @ibrahimkhairy 
config.tcl
```
export FASTROUTE_TCL = $(PLATFORM_DIR)/fastroute.tcl
export CELL_PAD_IN_SITES = 1
export CELL_PAD_IN_SITES_GLOBAL_PLACEMENT ?= 1
export CELL_PAD_IN_SITES_DETAIL_PLACEMENT ?= 1

export PLACE_DENSITY=0.4
```

./designs/$(PLATFORM)/coyote_tc/coyote_tc.package.strategy
```
 pad_inst_name "u_%s"
 pad_pin_name "%s"
 + filler_overlap_allowed 1
```
By updating above two files able to complete the flow. Please do confirm if that works at your end to make PR.

closed 2026-04-17 by github-actions[bot]