The-OpenROAD-Project/OpenROAD-flow-scripts #106: Prequisites/dependency handling

andreaskuster, 2021-07-28:
| Info | |
| --- | --- |
| OS | Ubuntu 20.04 LTS |
| OpenROAD-flow-scripts version | master (006e657571460a89492b0138d57c9118f2a20261) |
| Build option | --local |

**Is your feature request related to a problem? Please describe.**
Local installation using the `build_openroad.sh` script is more tedious than it has to be, if external dependencies (i.e. tcl-dev, libreadline-dev, ..) would either automatically be installed or listed in the README file. At the moment, the installation fails upon a missing dependency, and one has to re-run the script for each dependency (after installing it).

**Describe the solution you'd like**
I would integrate the dependency installation into the `build_openroad.sh` script. Alternatively, one could add it at least into the README description i.e.:

```
## prequisites (ubuntu 20.04)
sudo apt install tcl-dev git build-essential libreadline-dev libffi-dev swig libboost-all-dev libspdlog-dev libeigen3-dev mercurial klayout

hg clone http://lemon.cs.elte.hu/hg/lemon
cd lemon && mkdir build && cd build && cmake .. && make && make install && cd ../..
```

**Examples**
- CMake errors due to missing packages (see prequisites list, i.e. tcl-dev, libreadline-dev, ...)
- Missing KLayout issue arises during the flow execution:
Cryptic message for missing KLayout (` $(KLAYOUT_CMD)` variable is empty and therefore it triggers an error on the nested stdbuf tool). Furthermore, this should be handled by the `build_openroad.sh` script i.e. such that this script creates a fully functional eda toolflow environment.
```
andreas@andreas-vm:~/OpenROAD-flow-scripts/flow$ make
[INFO][FLOW] Using platform directory ./platforms/nangate45
(/usr/bin/time -f "%Eelapsed %PCPU %MmemKB" stdbuf -o L  -zz -rd design_name=gcd \
        -rd in_def=results/nangate45/gcd/base/6_final.def \
        -rd in_files="./platforms/nangate45/gds/NangateOpenCellLibrary.gds  " \
        -rd config_file= \
        -rd seal_file="" \
        -rd out_file=results/nangate45/gcd/base/6_1_merged.gds \
        -rd tech_file=./objects/nangate45/gcd/base/klayout.lyt \
        -rm ./util/def2stream.py) 2>&1 | tee ./logs/nangate45/gcd/base/6_1_merge.log
stdbuf: invalid option -- 'z'
Try 'stdbuf --help' for more information.
Command exited with non-zero status 125
0:00.00elapsed 0%CPU 2188memKB
make: *** [Makefile:614: results/nangate45/gcd/base/6_1_merged.gds] Error 125
```

maliberty, 2021-07-28:
Use etc/DependencyInstaller.sh to install all required dependencies for a local build.

maliberty, 2021-07-28:
@vvbandeira @dralabeing please make this clear in the docs.

andreaskuster, 2021-07-28:
> 
> 
> Use etc/DependencyInstaller.sh to install all required dependencies for a local build.

Thanks for the swift response. I can confirm that it works with `etc/DependencyInstaller.sh`. It would be great to add this to the README section `Building from Source`

dralabeing, 2021-07-28:
Will take care of this

rovinski, 2021-07-29:
The README does specify where to find the list of dependencies, but yes, a pointer to `tc/DependencyInstaller.sh` would be useful.

dralabeing, 2021-09-25:
This is resolved.

closed 2021-09-25 by dralabeing