The-OpenROAD-Project/OpenROAD-flow-scripts #124: Not able to run new design

vijayank88, 2021-08-19:
Its basic question. I ran new designs by adding into container path ./designs/sky130hd/newdesign last week. After git pull latest update, if I run new design by copying file to docker container its fail with  error; 
**bash-4.2$ make DESIGN_CONFIG=./designs/nangate45/eth_top/config.mk
[INFO][FLOW] Using platform directory ./platforms/nangate45
make: * platforms/nangate45: Is a directory. Stop.**

In-build design aes/riscv32i etc able to run successfully. using #make command

vvbandeira, 2021-08-19:
@vijayank88 could you provide as many as possible of the following:

- Test case (design source if public and `config.mk`) either a tarball or link to GitHub repository
- Git hash you are using
- Step-by-step to reproduce the error
- Complete log/commands output

Also, you first mentions `sky130hd` and the error mentions `nangate45`, possibly the issue is with your `config.mk`

vijayank88, 2021-08-20:
@vvbandeira  my problem is completely different.
OpenRoad inbuilt example working fine. Even if I create new project and src file within docker container able to proceed without any issue.
Only if I copy source file directory (RTL Codes) from local system to Docker I'm above error message.
Also if I see uid:gid is 1000:1000 all other files are in root:root.
Is that causing any issue?

vijayank88, 2021-08-20:
Just found it default shell is csh.. I switched to bash mode and used docker cp and there is no such error. Thanks

closed 2021-08-20 by vijayank88