The-OpenROAD-Project/OpenROAD-flow-scripts #146: depenndency installer failing

vijayank88, 2021-09-02:
I am trying to install open road flow locally. Trying to install dependencies for Centos7 with below command:
sudo ./etc/DependencyInstaller.sh -run
Got below error message:
No packages marked for update
Loaded plugins: fastestmirror, langpacks
epel-release-latest-7.noarch.rpm                                                                                                                                                            |  15 kB  00:00:00     
Examining /var/tmp/yum-root-jUgMLP/epel-release-latest-7.noarch.rpm: epel-release-7-13.noarch
/var/tmp/yum-root-jUgMLP/epel-release-latest-7.noarch.rpm: does not update installed package.
Error: Nothing to do

Looking for solution to fix above error.

uname -a: Linux hp11 3.10.0-1160.36.2.el7.x86_64 #1 SMP Wed Jul 21 11:57:15 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

vvbandeira, 2021-09-02:
Try to comment out the line that installs [epel](https://github.com/The-OpenROAD-Project/OpenROAD/blob/4d956498152d921d73c06ca161f90356d7bba3a5/etc/DependencyInstaller.sh#L152), looks like you already have it on your system. Then run the script again. Repeat if it complains about any other `rpm` package like [these](https://github.com/The-OpenROAD-Project/OpenROAD/blob/4d956498152d921d73c06ca161f90356d7bba3a5/etc/DependencyInstaller.sh#L125-L126).

vijayank88, 2021-09-03:
@vvbandeira  thanks for the update, able to install all dev packages successfully. 
I used #./build_openroad.sh --local after 94% completion of installation got below error:

[ 94%] ABC: `` Building binary: abc-341db25
make[1]: Leaving directory `/home/vijayan/OPEN_ROAD_FLOW/ORFS_LOCALL/OpenROAD-flow-scripts/tools/yosys/abc'
[100%] Building yosys-abc
mkdir -p /home/vijayan/OPEN_ROAD_FLOW/ORFS_LOCALL/OpenROAD-flow-scripts/tools/install/yosys/bin
cp yosys yosys-config yosys-abc yosys-filterlib yosys-smtbmc /home/vijayan/OPEN_ROAD_FLOW/ORFS_LOCALL/OpenROAD-flow-scripts/tools/install/yosys/bin
strip -S /home/vijayan/OPEN_ROAD_FLOW/ORFS_LOCALL/OpenROAD-flow-scripts/tools/install/yosys/bin/yosys
strip /home/vijayan/OPEN_ROAD_FLOW/ORFS_LOCALL/OpenROAD-flow-scripts/tools/install/yosys/bin/yosys-abc
strip /home/vijayan/OPEN_ROAD_FLOW/ORFS_LOCALL/OpenROAD-flow-scripts/tools/install/yosys/bin/yosys-filterlib
mkdir -p /home/vijayan/OPEN_ROAD_FLOW/ORFS_LOCALL/OpenROAD-flow-scripts/tools/install/yosys/share/yosys
cp -r share/. /home/vijayan/OPEN_ROAD_FLOW/ORFS_LOCALL/OpenROAD-flow-scripts/tools/install/yosys/share/yosys/.
make: Leaving directory `/home/vijayan/OPEN_ROAD_FLOW/ORFS_LOCALL/OpenROAD-flow-scripts/tools/yosys'
-- The CXX compiler identification is GNU 4.8.5
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: 1
-- OpenROAD git sha: 4dca404c76b3302aa4e6ef8af1cd976c8de91cb4
-- System name: Linux
-- Compiler: GNU 4.8.5
-- Build type: RELEASE
-- Install prefix: /home/vijayan/OPEN_ROAD_FLOW/ORFS_LOCALL/OpenROAD-flow-scripts/tools/install/OpenROAD
-- TCL library: /usr/lib64/libtcl.so
-- TCL header: /usr/include/tcl.h
-- TCL readline library: /usr/lib64/libtclreadline.so
-- TCL readline header: /usr/include
-- Found SWIG: /usr/bin/swig (found suitable version "4.0.1", minimum required is "3.0") 
-- Found Boost: /usr/local/lib/cmake/Boost-1.72.0/BoostConfig.cmake (found version "1.72.0")  
-- Found Python3: /usr/lib64/libpython3.6m.so (found version "3.6.8") found components: Development 
-- Found ZLIB: /usr/lib64/libz.so (found version "1.2.7") 
-- Looking for C++ include pthread.h
CMake Error in /home/vijayan/OPEN_ROAD_FLOW/ORFS_LOCALL/OpenROAD-flow-scripts/tools/OpenROAD/build/CMakeFiles/CMakeTmp/CMakeLists.txt:
  Target "cmTC_d1434" requires the language dialect "CXX17" , but CMake does
  not know the compile flags to use to enable it.

CMake Error at /usr/share/cmake3/Modules/CheckIncludeFileCXX.cmake:92 (try_compile):
  Failed to generate test project build system.
Call Stack (most recent call first):
  /usr/share/cmake3/Modules/FindThreads.cmake:144 (CHECK_INCLUDE_FILE_CXX)
  /usr/local/lib64/cmake/spdlog/spdlogConfig.cmake:4 (find_package)
  src/CMakeLists.txt:107 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/vijayan/OPEN_ROAD_FLOW/ORFS_LOCALL/OpenROAD-flow-scripts/tools/OpenROAD/build/CMakeFiles/CMakeOutput.log".

maliberty, 2021-09-03:
This is an ancient version:
  The CXX compiler identification is GNU 4.8.5 

you'll need something more recent.  You may need to do:

source scl_source enable devtoolset-8

before running cmake.

vijayank88, 2021-09-04:
@maliberty  @vvbandeira  thanks for the help. Installed successfully after few tweaking. Also below things can be updated in README.md helpful for new user if they install locally. (Optional Info user allowed to edit)
If user gets error message "Error: Nothing to do" skip those lines from ./etc/DependencyInstaller.sh and continue run.
source below files at the time of running installation.
source scl_source enable devtoolset-8
source scl_source enable llvm-toolset-7.0

closed 2021-09-04 by vijayank88