The-OpenROAD-Project/OpenROAD-flow-scripts #150: Unable to create a local build on CentOS7

paulgutwin, 2021-09-10:
**Describe the bug**
I am unable to successfully complete a local build. After running the dependency installer with the -dev option, I then run "./build_openroad.sh --local" but run into an issue with CMAKE including pthreads. I am not familiar with CMAKE and unable to debug the problem further.

As an aside, I am able to successfully build from the OpenROAD repo. 

**Expected behavior**
Successful local build.

**Screenshots**
.
.
.
make: Leaving directory `/home/pgutwin/OpenROAD-flow-scripts/tools/yosys'
-- The CXX compiler identification is GNU 4.8.5
-- Check for working CXX compiler: /bin/c++
-- Check for working CXX compiler: /bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: 1
-- OpenROAD git describe: v2.0-880-gd1c7001
-- System name: Linux
-- Compiler: GNU 4.8.5
-- Build type: RELEASE
-- Install prefix: /home/pgutwin/OpenROAD-flow-scripts/tools/install/OpenROAD
-- TCL library: /usr/lib64/libtcl.so
-- TCL header: /usr/include/tcl.h
-- TCL readline library: /usr/lib64/libtclreadline.so
-- TCL readline header: /usr/include
-- Found SWIG: /bin/swig (found suitable version "4.0.1", minimum required is "3.0") 
-- Boost  found.
-- Found Python3: /usr/lib64/libpython3.6m.so (found version "3.6.8") found components:  Development 
-- Found ZLIB: /usr/lib64/libz.so (found version "1.2.7") 
-- Looking for C++ include pthread.h
CMake Error in /home/pgutwin/OpenROAD-flow-scripts/tools/OpenROAD/build/CMakeFiles/CMakeTmp/CMakeLists.txt:
  Target "cmTC_66fb0" requires the language dialect "CXX17" , but CMake does
  not know the compile flags to use to enable it.

CMake Error at /usr/local/share/cmake-3.14/Modules/CheckIncludeFileCXX.cmake:92 (try_compile):
  Failed to generate test project build system.
Call Stack (most recent call first):
  /usr/local/share/cmake-3.14/Modules/FindThreads.cmake:110 (CHECK_INCLUDE_FILE_CXX)
  /usr/local/lib64/cmake/spdlog/spdlogConfig.cmake:4 (find_package)
  src/CMakeLists.txt:107 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/pgutwin/OpenROAD-flow-scripts/tools/OpenROAD/build/CMakeFiles/CMakeOutput.log".
[pgutwin@usvmchalx1 OpenROAD-flow-scripts]$ 

**Environment (please complete the following information):**
 - OS: CentOS 7
 - OpenROAD-flow commit eceb503

**Additional context**
I am running CentOS on a virtual machine starting with a minimal CentOS7 installation.

rovinski, 2021-09-10:
Your compiler is identified as GCC 4 whereas 7+ is required. See https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/146#issuecomment-912586947.

paulgutwin, 2021-09-10:
Hello Austin - thank you very much for the pointer.  I (stupidly) overlooked the need to enable GCC 8 and CLANG 7.  My build is now successful.

closed 2021-09-10 by paulgutwin

vvbandeira, 2021-09-11:
@paulgutwin, happy to know that you were able to build. Out of curiosity, during your first try, did you ever got to our [Getting Started](https://openroad.readthedocs.io/en/latest/user/GettingStarted.html) docs page and was it clear? Just asking to know if there's anything we can clarify in the docs to make things better. Thanks.

rovinski, 2021-09-11:
It is possible to give better error messages with something like
```
if (CMAKE_COMPILER_IS_GNUCC AND CMAKE_CXX_COMPILER_VERSION VERSION_LESS 7)
    message(FATAL_ERROR "OpenROAD requires GCC 7 or greater.")
endif()
```
But that is not as robust as `CXX_STANDARD_REQUIRED` which we currently use.

vvbandeira, 2021-09-13:
@rovinski can we use both? check first for the version then for std. req.?

paulgutwin, 2021-09-13:
@vvbandeira  yes, I was using the [Getting Started](https://openroad.readthedocs.io/en/latest/user/GettingStarted.html). However, just before the warning regarding enabling GCC-8/CLANG-7, there is a link [Build from sources locally](https://openroad.readthedocs.io/en/latest/user/BuildLocally.html) which of course I followed. Since that page also has instructions regarding verifying the build, I didn't think to go back to the "Getting Started" page. 

I think that my issue would have been avoided had the warning been repeated on the "Build from sources locally" page.