The-OpenROAD-Project/OpenROAD-flow-scripts #161: Cannot run a sample design coyote_tc in docker

peijunh, 2021-09-20:
**Describe the bug**
A clear and concise description of what the bug is. Please include any tags that apply to this issue:
I try to run the sample design sky130hd/coyote_tc in OpenROAD-flow-scripts inside docker. But I get an error at the beginning of floorplan. 

* [Design] for example design issues (i.e. the design does not pass functional validation, etc.)

**Expected behavior**
The design pass the whole flow without drc lvs error.

**Screenshots**
If applicable, add screenshots to help explain your problem (e.g. layout screenshots).
```
[WARNING ODB-0176] error: undefined layer (nwell) referenced
[INFO ODB-0225]     Created 1 library cells
[INFO ODB-0226] Finished LEF file:  ./platforms/sky130io/lef/sky130_ef_io__vssio_hvc_pad.lef
[INFO ODB-0222] Reading LEF file: ./platforms/sky130io/lef/sky130_ef_io__vssio_lvc_pad.lef
[WARNING ODB-0220] WARNING (LEFPARS-2008): NOWIREEXTENSIONATPIN statement is obsolete in version 5.6 or later.
The NOWIREEXTENSIONATPIN statement will be ignored. See file ./platforms/sky130io/lef/sky130_ef_io__vssio_lvc_pad.lef at line 2.

[INFO ODB-0225]     Created 1 library cells
[INFO ODB-0226] Finished LEF file:  ./platforms/sky130io/lef/sky130_ef_io__vssio_lvc_pad.lef
[INFO ODB-0222] Reading LEF file: ./platforms/sky130ram/sky130_sram_1rw1r_80x64_8/sky130_sram_1rw1r_80x64_8.lef
[INFO ODB-0225]     Created 1 library cells
[INFO ODB-0226] Finished LEF file:  ./platforms/sky130ram/sky130_sram_1rw1r_80x64_8/sky130_sram_1rw1r_80x64_8.lef
[INFO ODB-0222] Reading LEF file: ./platforms/sky130ram/sky130_sram_1rw1r_128x256_8/sky130_sram_1rw1r_128x256_8.lef
[INFO ODB-0221] 1000000 lines parsed!
[INFO ODB-0225]     Created 1 library cells
[INFO ODB-0226] Finished LEF file:  ./platforms/sky130ram/sky130_sram_1rw1r_128x256_8/sky130_sram_1rw1r_128x256_8.lef
[INFO ODB-0222] Reading LEF file: ./platforms/sky130ram/sky130_sram_1rw1r_44x64_8/sky130_sram_1rw1r_44x64_8.lef
[INFO ODB-0225]     Created 1 library cells
[INFO ODB-0226] Finished LEF file:  ./platforms/sky130ram/sky130_sram_1rw1r_44x64_8/sky130_sram_1rw1r_44x64_8.lef
[INFO ODB-0222] Reading LEF file: ./platforms/sky130ram/sky130_sram_1rw1r_64x256_8/sky130_sram_1rw1r_64x256_8.lef
[INFO ODB-0221] 1000000 lines parsed!
[INFO ODB-0225]     Created 1 library cells
[INFO ODB-0226] Finished LEF file:  ./platforms/sky130ram/sky130_sram_1rw1r_64x256_8/sky130_sram_1rw1r_64x256_8.lef
[WARNING ORD-1011] LEF master sky130_ef_io__gpiov2_pad_wrapped has no liberty cell.
[WARNING STA-0337] port 'en_i' not found.
[WARNING STA-0337] port 'en_i' not found.
[WARNING STA-0337] port 'en_i' not found.
number instances in verilog is 197690
Loading library IO cell meta-data
Loaded library IO cell data
[INFO IFP-0001] Added 1536 rows of 10391 sites.
[WARNING PDN-0035] No track information found for layer li1
[WARNING PDN-0035] No track information found for layer met1
[WARNING PDN-0035] No track information found for layer met2
[WARNING PDN-0035] No track information found for layer met3
[WARNING PDN-0035] No track information found for layer met4
[WARNING PDN-0035] No track information found for layer met5
Error: floorplan.tcl, 59 key "name" not known in dictionary
0:10.75elapsed 99%CPU 911708memKB
make: *** [results/sky130hd/coyote_tc/base/2_1_floorplan.def] Error 1
```
**Environment (please complete the following information):**
Please only report issues for supported OSs.
 - OS: CentOS 7
 - OpenROAD-flow commit 99889aac8cfbe6c96da2001e43f71d4522f144e2

**File Uploads**

**Additional context**
I run the design in docker. I change the Makefile to build coyote_tc. The command is as follows:
>git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts
>cd OpenROAD-flow-scripts
>./build_openroad.sh
>docker run -it -u $(id -u ${USER}):$(id -g ${USER}) -v $(pwd)/flow/platforms:/OpenROAD-flow-scripts/flow/platforms:ro openroad/flow-scripts
>source ./setup_env.sh
>cd flow
>make

maliberty, 2021-09-20:
Would you try it with:

./build_openroad.sh --latest

vvbandeira, 2021-09-20:
I don't think this design was fully ported into ORFS for public PKDs. There are two stale branches (one from Colin and one I tried to rebase his into master) on the private org, neither work last time I checked.

peijunh, 2021-09-20:
@maliberty Thanks! I tried this before I run another flow of OpenROAD-flow-scripts. I got the same floorplan error. @vvbandeira is there a branch where this design work?

vvbandeira, 2021-09-20:
@peijunh not that I am aware. If someone wants to try to make the design work and have access to the private org, take a look at branches: coyote-tc-nightly and sky130_coyote_tc.

msaligane, 2021-09-20:
@peijunh Let's switch focus on other designs that works first then we could loop back to this one.

vijayank88, 2021-09-29:
I'm also tried. Problem comes during initialize floor plan with reading site lef file. 
Error message as follows with latest OR commit.

[INFO IFP-0001] Added 1535 rows of 10390 sites.
[ERROR PAD-0181] Library cell reference vssio_lvc_ground_pad missing type attribute.
Error: floorplan.tcl, 59 PAD-0181
Command exited with non-zero status 1
Elapsed time: 0:28.44[h:]min:sec. Average CPU: 68%. Peak memory: 868404KB.
make: *** [Makefile:387: results/sky130hd/coyote_tc/base/2_1_floorplan.def] Error 1

maliberty, 2021-09-29:
This design got missed from our all designs list and so was not being tested.  We will take a look at fixing it but I suggest you try another for now.  Sorry for the bother.

closed 2023-01-10 by tspyrou