The-OpenROAD-Project/OpenROAD-flow-scripts #164: chameleon_hier struck at GRT stage

vijayank88, 2021-09-25:
**Describe the bug**
chameleon_hier I'm trying to run in OpenRoad-flow-scripts using sky130hd tech and enabling  DESIGN_CONFIG=./designs/sky130hd/chameleon_hier/config.mk in Makefile using make command in terminal.
But its got stop at global routing stage with below message:
 [ERROR GRT-0118] Routing congestion too high.
Error: global_route.tcl, 42 GRT-0118

Its in-built example donot know why its stops with this error. Looking for the workaround.

**Expected behavior**
complete the flow without error

**Environment (please complete the following information):**
Please only report issues for supported OSs.
 - OS: CentOS 7
 - OpenROAD-flow version: v2.0-1020-g01461c345

**File Uploads**

[INFO GRT-0106] Layer assignment begins.
[INFO GRT-0107] Layer assignment finished.
[INFO GRT-0197] Via related to pin nodes: 15638
[INFO GRT-0198] Via related Steiner nodes: 315
[INFO GRT-0199] Via filling finished.
[INFO GRT-0111] Final number of vias: 21784
[INFO GRT-0112] Final usage 3D: 287715
[WARNING GRT-0211] dbGcellGrid already exists in db. Clearing existing dbGCellGrid.
[ERROR GRT-0118] Routing congestion too high.
Error: global_route.tcl, 42 GRT-0118
Elapsed time: 2:48.47[h:]min:sec. Average CPU: 99%. Peak memory: 2817924KB.
make: *** [Makefile:535: results/sky130hd/chameleon_hier/base/route.guide] Error 1

vvbandeira, 2021-12-13:
@vijayank88 can you test again with the latest code?

vijayank88, 2021-12-14:
@vvbandeira With latest update congestion issue gone. But during final report generation got below error message
```
[INFO PSM-0031] Number of PDN nodes on net VDD = 2554894.
[INFO PSM-0064] Number of voltage sources = 345.
[INFO PSM-0040] All PDN stripes on net VDD are connected.
[ERROR PSM-0010] LU factorization of the G Matrix failed.
Error: final_report.tcl, 81 PSM-0010
Elapsed time: 4:59.61[h:]min:sec. CPU time: user 295.19 sys 4.10 (99%). Peak memory: 5372804KB.
```

maliberty, 2021-12-14:
This was recently fixed - are you using the latest OR?

vijayank88, 2021-12-20:
OpenROAD v2.0-1978-gc1af9a878 
The above Openroad flow scripts tag able to complete the flow. Closing this issue.

closed 2021-12-20 by vijayank88