The-OpenROAD-Project/OpenROAD-flow-scripts #165: Unable to build from sources using docker

akashpatil98, 2021-09-25:
**Describe the bug**
I'm trying to follow the steps mentioned in the [OpenROAD documentation](https://openroad.readthedocs.io/en/latest/user/BuildWithDocker.html) for **Building from sources using docker** option. 

I was able to successfully follow the steps mentioned in the guide till **Clone and Build** option. 
In the final step: **Verify Installation**
I'm get an unexpected output when I run the command:
``` console
sudo docker run -it -u $(id -u ${USER}):$(id -g ${USER}) -v $(pwd)/flow/platforms:/OpenROAD-flow-scripts/flow/platforms:ro openroad/flow-scripts
```
**Terminal Output - After running the above command**
```console
akash@akash-lenovo:~/OpenROAD-flow-scripts$ sudo docker run -it -u $(id -u ${USER}):$(id -g ${USER}) -v $(pwd)/flow/platforms:/OpenROAD-flow-scripts/flow/platforms:ro openroad/flow-scripts
[sudo] password for akash: 
Unable to find image 'openroad/flow-scripts:latest' locally
docker: Error response from daemon: pull access denied for openroad/flow-scripts, repository does not exist or may require 'docker login': denied: requested access to the resource is denied.
See 'docker run --help'.
akash@akash-lenovo:~/OpenROAD-flow-scripts$ 

```
**Screenshots**
[image]

**Environment:**
 - OS: Ubuntu 18.04.6
 - Docker: Version 20.10.8, build 3967b7d

vvbandeira, 2021-09-26:
@akashpatil98 are you using the latest master? Also, what is the output of `./build_openroad.sh`?

akashpatil98, 2021-09-26:
@vvbandeira Yes, I'm using the latest master. The commit is 858f67d1e2509a04e9d2f085cebf6017446bee8c
To reconfirm that I'm using the latest master, I also checked `git status -uno` and this is the output:
```console
akash@akash-lenovo:~/OpenROAD-flow-scripts$ git status -uno
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit (use -u to show untracked files)
akash@akash-lenovo:~/OpenROAD-flow-scripts$ 
```
**Output of ./build_openroad.sh**
Please find the complete console output below:

P.S. - I've added the complete output thinking it might be easier to debug. I apologize if I wasn't supposed to add the complete console output after running `./build_openroad.sh`.

```console
akash@akash-lenovo:~/OpenROAD-flow-scripts$ sudo ./build_openroad.sh
[sudo] password for akash: 
[INFO FLW-0000] Using docker build method. This will create a docker image tagged 'flow-scripts'.
Sending build context to Docker daemon  11.07MB
Step 1/4 : FROM openroad/yosys-dev AS builder
 ---> 2842cd8a1332
Step 2/4 : COPY . /yosys
 ---> 0d75ccfccd6a
Step 3/4 : WORKDIR /yosys
 ---> Running in ca31bbc915ab
Removing intermediate container ca31bbc915ab
 ---> c431ae412a07
Step 4/4 : RUN make PREFIX=/install CONFIG=gcc ABCREV=bafd2a7 ABCURL=https://github.com/berkeley-abc/abc install -j$(nproc)
 ---> Running in 9adf43590ecd
[  0%] Building kernel/version_UNKNOWN.cc
[  0%] Building kernel/driver.o
[  0%] Building techlibs/common/simlib_help.inc
[  0%] Building techlibs/common/simcells_help.inc
[  1%] Building kernel/rtlil.o
[  1%] Building kernel/log.o
[  2%] Building kernel/calc.o
[  2%] Building kernel/yosys.o
[  3%] Building kernel/cellaigs.o
[  3%] Building kernel/celledges.o
[  3%] Building kernel/satgen.o
[  4%] Building kernel/mem.o
[  4%] Building libs/bigint/BigIntegerAlgorithms.o
[  4%] Building libs/bigint/BigInteger.o
[  5%] Building libs/bigint/BigIntegerUtils.o
[  5%] Building libs/bigint/BigUnsigned.o
[  6%] Building libs/bigint/BigUnsignedInABase.o
[  6%] Building libs/sha1/sha1.o
[  6%] Building libs/json11/json11.o
[  7%] Building libs/subcircuit/subcircuit.o
[  7%] Building libs/ezsat/ezsat.o
kernel/satgen.cc: In member function 'bool Yosys::SatGen::importCell(Yosys::RTLIL::Cell*, int)':
kernel/satgen.cc:1137:32: warning: 'undef_srst' may be used uninitialized in this function [-Wmaybe-uninitialized]
      std::tie(d, undef_d) = mux(srst, undef_srst, rval, undef_rval, d, undef_d);
                             ~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
kernel/satgen.cc:1123:32: warning: 'undef_en' may be used uninitialized in this function [-Wmaybe-uninitialized]
      std::tie(d, undef_d) = mux(en, undef_en, d, undef_d, old_q, undef_old_q);
                             ~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
kernel/satgen.cc:1109:32: warning: 'undef_srst' may be used uninitialized in this function [-Wmaybe-uninitialized]
      std::tie(d, undef_d) = mux(srst, undef_srst, rval, undef_rval, d, undef_d);
                             ~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[  7%] Building libs/ezsat/ezminisat.o
[  8%] Building libs/minisat/Options.o
[  8%] Building libs/minisat/SimpSolver.o
[  9%] Building libs/minisat/Solver.o
[  9%] Building libs/minisat/System.o
[  9%] Building frontends/aiger/aigerparse.o
[ 10%] Building frontends/rpc/rpc_frontend.o
[ 10%] Building frontends/verific/verific.o
[ 10%] Building frontends/json/jsonparse.o
[... 2811 lines cut ...]
Step 8/10 : COPY ./setup_env.sh .
 ---> 032ba1e53383
Step 9/10 : COPY ./flow ./flow
 ---> cb891b4f8d43
Step 10/10 : RUN chmod o+rw -R /OpenROAD-flow-scripts
 ---> Running in 6e14d552d3f9
Removing intermediate container 6e14d552d3f9
 ---> e46cf80e662c
Successfully built e46cf80e662c
Successfully tagged flow-scripts:latest
akash@akash-lenovo:~/OpenROAD-flow-scripts$ 
```
**Running the next command**
```console
sudo docker run -it -u $(id -u ${USER}):$(id -g ${USER}) -v $(pwd)/flow/platforms:/OpenROAD-flow-scripts/flow/platforms:ro openroad/flow-scripts
```
**Console output after running the above command**
```console
akash@akash-lenovo:~/OpenROAD-flow-scripts$ sudo docker run -it -u $(id -u ${USER}):$(id -g ${USER}) -v $(pwd)/flow/platforms:/OpenROAD-flow-scripts/flow/platforms:ro openroad/flow-scripts
[sudo] password for akash: 
Unable to find image 'openroad/flow-scripts:latest' locally
docker: Error response from daemon: pull access denied for openroad/flow-scripts, repository does not exist or may require 'docker login': denied: requested access to the resource is denied.
See 'docker run --help'.
akash@akash-lenovo:~/OpenROAD-flow-scripts$ 
```

vvbandeira, 2021-09-26:
@akashpatil98 please try:
```
docker run -it -u $(id -u ${USER}):$(id -g ${USER}) -v $(pwd)/flow/platforms:/OpenROAD-flow-scripts/flow/platforms:ro flow-scripts:latest
```

akashpatil98, 2021-09-26:
@vvbandeira Thank you very much! It worked.

I was able to complete the remaining steps mentioned in the [documentation](https://openroad.readthedocs.io/en/latest/user/BuildWithDocker.html)

**Solved: Console output after running the command successfully**
```console
akash@akash-lenovo:~/OpenROAD-flow-scripts$ sudo docker run -it -u $(id -u ${USER}):$(id -g ${USER}) -v $(pwd)/flow/platforms:/OpenROAD-flow-scripts/flow/platforms:ro flow-scripts:latest
[sudo] password for akash: 
bash-4.2$ source ./setup_env.sh
OPENROAD: /OpenROAD-flow-scripts/tools/OpenROAD
bash-4.2$ yosys -help

Usage: yosys [options] [<infile> [..]]

    -Q
        suppress printing of banner (copyright, disclaimer, version)

    -T
        suppress printing of footer (log hash, version, timing statistics)

    -q
        quiet operation. only write warnings and error messages to console
        use this option twice to also quiet warning messages

    -v <level>
        print log headers up to level <level> to the console. (this
        implies -q for everything except the 'End of script.' message.)

    -t
        annotate all log messages with a time stamp

    -d
        print more detailed timing stats at exit

    -l logfile
        write log messages to the specified file

    -L logfile
        like -l but open log file in line buffered mode

    -o outfile
        write the design to the specified file on exit

    -b backend
        use this backend for the output file specified on the command line

    -f frontend
        use the specified frontend for the input files on the command line

    -H
        print the command list

    -h command
        print the help message for the specified command

    -s scriptfile
        execute the commands in the script file

    -c tcl_scriptfile
        execute the commands in the tcl script file (see 'help tcl' for details)

    -p command
        execute the commands

    -m module_file
        load the specified module (aka plugin)

    -X
        enable tracing of core data structure changes. for debugging

    -M
        will slightly randomize allocated pointer addresses. for debugging

    -A
        will call abort() at the end of the script. for debugging

    -D <macro>[=<value>]
        set the specified Verilog define (via "read -define")

    -P <header_id>[:<filename>]
        dump the design when printing the specified log header to a file.
        yosys_dump_<header_id>.il is used as filename if none is specified.
        Use 'ALL' as <header_id> to dump at every header.

    -W regex
        print a warning for all log messages matching the regex.

    -w regex
        if a warning message matches the regex, it is printed as regular
        message instead.

    -e regex
        if a warning message matches the regex, it is printed as error
        message instead and the tool terminates with a nonzero return code.

    -E <depsfile>
        write a Makefile dependencies file with in- and output file names

    -x <feature>
        do not print warnings for the specified experimental feature

    -g
        globally enable debug log messages

    -V
        print version information and exit

The option -S is an shortcut for calling the "synth" command, a default
script for transforming the Verilog input to a gate-level netlist. For example:

    yosys -o output.blif -S input.v

For more complex synthesis jobs it is recommended to use the read_* and write_*
commands in a script file instead of specifying input and output files on the
command line.

When no commands, script files or input files are specified on the command
line, yosys automatically enters the interactive command mode. Use the 'help'
command to get information on the individual commands.
```

closed 2021-09-26 by akashpatil98