The-OpenROAD-Project/OpenROAD-flow-scripts #173: 'Routing Congestion too high' error at Global Routing step

lakshmi-sathi, 2021-10-01:
**Describe the bug**
The flow fails for this BlackParrot front-end design ([bp_fe](https://github.com/bsg-idea/bsg_sky130_designs/tree/test_sky130/designs)) at the Global Routing step with "Routing congestion too high" error. This issue is despite providing a large core area of 11020x8120um. It is using fakerams generated using [bsg_fakeram generator](https://github.com/bespoke-silicon-group/bsg_fakeram) as placeholders for its memories. The same routing congestion issue is noted on another BlackParrot design too which uses fakeram macros like this one.
The logs of the run for this design are available here: https://github.com/bsg-idea/bsg_sky130_designs/tree/test_sky130/designs/bp_fe/logs

**Expected behavior**
Routing succeeding without routing congestion or violations.

**Environment**
 - OS: Ubuntu 20.04
 - OpenROAD-flow v2.0-880-gd1c7001ad

**File Uploads**
Just have to place this folder in the 'designs/sky130hd/' directory and run it as usual for reproducing the issue: https://github.com/bsg-idea/bsg_sky130_designs/tree/test_sky130/designs/bp_fe

**Additional context**
The congestion issue does not seem to appear on designs without SRAM macros. Different pin widths, pin spacings, and sizes were tried (both large and small) for the SRAM macros but still, the issue persists. [bsg_cache_dma](https://github.com/bsg-idea/bsg_sky130_designs/tree/test_sky130/designs/bsg_cache_dma_mem) is a design with a fakeram for which the OpenROAD flow was successful (the OpenLane flow failed for this bsg_cache_dma design with the routing congestion issue)

@tspyrou  @taylor-bsg

maliberty, 2021-10-01:
When you have macros you can still have congestion issues in the channels no matter the core size.    Have you tested increasing MACRO_PLACE_HALO and MACRO_PLACE_CHANNEL?

tspyrou, 2021-10-01:
@lakshmi-sathi please try Matt's suggestion and close if it works or assign to @eder-matheus  if it doesn't.

lakshmi-sathi, 2021-10-02:
@maliberty Thanks for that pointer. I tried it out. 
Originally the values were:
```
export MACRO_PLACE_HALO ?= 1 1
export MACRO_PLACE_CHANNEL ?= 80 80
```
I changed it to (I am editing directly in the platform/sky130hd/config.mk):
```
export MACRO_PLACE_HALO ?= 3 3
export MACRO_PLACE_CHANNEL ?= 100 100
```
Now for some reason it appears to do worse (the overflow values seem larger) and then it terminates with this error: 

[image]

It doesn't seem to be a memory issue as 30GB of RAM is available on the system.

vijayank88, 2021-10-04:
@lakshmi-sathi  seems you're trying 45nm tech constraint to 130nm node directly. Maybe slight modification required in config and constraint file as well to avoid congestion issue.

vijayank88, 2021-10-05:
@lakshmi-sathi Its seems you just copied nangate45 tech example and trying to use sky130nm technology for GDSII generation.
Have you modified configuration file accordingly?

lakshmi-sathi, 2021-10-05:
@vijayank88 When I collected the files again for the design I must have
missed it, thanks for pointing that out. I had tried giving custom
constraints before but it hadn't solved the issue. Now after I edited the
constraints I gave it another try but the issue remains the same.
Might you know maybe, of any options that I could try modifying to tackle
this congestion issue? ( I have tried density, area, cell pad sites, macro
place halo and macro place channel.)

lakshmi-sathi, 2021-10-05:
> Hi Lakshmi,
>
> If you have any fixes, can you be sure to update in the repo so others can
> work off of the latest?
>
> M

Yes Prof., I will keep the designs in the repo upto date with the changes
that I make, and if I am able to get an improved result, I will update the
issue with the same.

lakshmi-sathi, 2021-10-12:
@eder-matheus , Hi
  Two other designs from issue #174 could get through the flow by adjusting the MACRO_PLACE_CHANNEL variable so I tried adjusting the MACRO_PLACE_HALO and MACRO_PLACE_CHANNEL variables some more and I was trying to make sense of what I observed:
  Increasing the MACRO_PLACE_HALO or MACRO_PLACE_CHANNEL variables reduces the congestion but increasing beyond a point strangely only worsens the congestion (If we increase it shouldn't be causing congestion, am I right?). At a sweet spot of MACRO_PLACE_HALO = 100 100  & MACRO_PLACE_CHANNEL = 200 200, I am able to get the lowest congestion with a final overflow of 85 (Image below)(I have updated this in the repo link given in the main comment)
  
[image]

This is however with a very low place density of 0.03. Any place density larger than that is increasing the congestion. Could you please take a look and give me an idea what I could be missing here. I am attaching the packed results:
[global_route_issue.tar.gz](https://drive.google.com/file/d/1rHqVN-AY_8-uBf1zHE2lGKnWMaqfwl5q/view?usp=sharing)

eder-matheus, 2021-10-12:
Hi, @lakshmi-sathi. Here is the congestion map of your design after the congestion error:
[image]
The red regions are the ones with overflow, and they are all over the macros. Is it possible that you have unreachable pins on these macros?

maliberty, 2021-10-12:
@eder-matheus the macros themselves will likely have blockages producing the red color.  The red around the edges of the macros is more interesting.

rovinski, 2021-10-13:
> The macro LEFs (fakerams) are made of metal layers 1,2,3 and 4 so the router should be able to route over the macros by default (using metal 5), am I right?

Only if the macro cell power grid does not block a significant number of routes on M5, and M5 is a horizontal layer. If not, then that's a good question.

> And if routing over the macro is possible then the macro orientation should not be that much an issue, isn't it?

Regardless of if it's _possible_, if the connectivity is much stronger to the core than the I/O, it is much more beneficial to have the macro flipped in order to reduce wirelength.

taylor-bsg, 2021-10-15:
@eder-matheus What is the issue in the commercial tool? Are there too many pins on the periphery of the design, too many pins on the RAM, .. or?  If M5 is free, then it seems like the ram itself should not present a significant blockage.

eder-matheus, 2021-11-10:
@lakshmi-sathi Sorry for my delay on it. Could you try removing the line `set_macro_extension 1` in file `platforms/sky130hd/fastroute.tcl`? It is creating extra blockages around the macros that makes impossible for the router to access the pins. This workaround works for Issue https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/174 too.

eder-matheus, 2021-11-16:
@lakshmi-sathi The PR https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/220 should fix the issue you're facing when running the flow. Could you try it?

maliberty, 2021-11-24:
@lakshmi-sathi have you tried it?

lakshmi-sathi, 2021-11-24:
Sorry about the delay. I missed the previous comment. I was collecting together the different designs with which I had faced this issue.
I verified this with 4 different designs (includes designs with fakerams, realrams and synthesized rams) that were facing this issue previously and I can confirm that removing `set_macro_extension 1` is solving this congestion issue. Could you share what exactly is the purpose of this parameter, is it used to create blockages around the macros for some purpose?

I had tried to push quite a few different designs with sky130 SRAMs through the flow but in almost all cases (all major cases) the flow failed with congestion too high error. Only some smaller designs like tinyRocket and Serv passed the flow before this fix (and that was by adjusting the MACRO_PLACE_CHANNEL). 
Before the fix, I had tried several different things, like:
* Macro pin spacing
* Macro pin width
* Macro Halo (spacing between standard cells and macro)
* Macro channel (spacing between macros)
* Place density
* Utilization / Core area
* Tried fake and real SRAMs
* Manual orientation of macros.
* Manually placing just 4 macros per row spaced out maximally (Image below)
* Reducing design I/O by cutting down top levels
* Increasing the power grid spacing (Image below)
* And more
But none had solved this congestion issue. 

I have tested this fix on the BlackParrot front end design too and it passes the flow.

maliberty, 2021-11-24:
Macro extension is mostly used on more advanced nodes to avoid routing too close to macros where restrictive rules can make routing difficult.  It wasn't used on sky130 until recently we added it due to a poor RAM LEF in the Chameleon design.  Once we realized the issue we moved it from the platform level to that specific design.  I'm glad things are much better now.

Is there anything further to resolve here or should we close this?

lakshmi-sathi, 2021-11-27:
After the fix, the 'routing congestion too high' issue at global routing is not faced. I have linked this issue with #240, so that it will get closed automatically.

mousaq92, 2022-02-07:
> **Describe the bug** The flow fails for this BlackParrot front-end design ([bp_fe](https://github.com/bsg-idea/bsg_sky130_designs/tree/test_sky130/designs)) at the Global Routing step with "Routing congestion too high" error. This issue is despite providing a large core area of 11020x8120um. It is using fakerams generated using [bsg_fakeram generator](https://github.com/bespoke-silicon-group/bsg_fakeram) as placeholders for its memories. The same routing congestion issue is noted on another BlackParrot design too which uses fakeram macros like this one. The logs of the run for this design are placed here: https://github.com/bsg-idea/bsg_sky130_designs/tree/test_sky130/designs/bp_fe/logs
> 
> **Expected behavior** Routing succeeding without routing congestion or violations.
> 
> **Environment**
> 
> * OS: Ubuntu 20.04
> * OpenROAD-flow v2.0-880-gd1c7001ad
> 
> **File Uploads** Just have to place this folder in the 'designs/sky130hd/' directory and run it as usual for reproducing the issue: https://github.com/bsg-idea/bsg_sky130_designs/tree/test_sky130/designs/bp_fe
> 
> **Additional context** The congestion issue does not seem to appear on designs without SRAM macros. Different pin widths, pin spacings, and sizes were tried (both large and small) for the fakerams but still, the issue persists. [bsg_cache_dma](https://github.com/bsg-idea/bsg_sky130_designs/tree/test_sky130/designs/bsg_cache_dma_mem) is a design with a fakeram for which the OpenROAD flow was successful.
> 
> @tspyrou @taylor-bsg

I have the same issue, except I am not using any macros in my design. Routing congestion too high.

maliberty, 2022-02-14:
@mousaq92 that sounds like an unrelated issue then and should have a separate issue with a reproducer.

Ali-Sabir2, 2022-08-16:
Hi!
I am facing routing congestion issue.
Can anyone help me ? 
How to solve this problem?
Thanks

Ali-Sabir2, 2022-08-16:
[image]

vijayank88, 2022-08-16:
@Ali-Sabir2 
Please open your issue here: https://github.com/The-OpenROAD-Project/OpenLane/issues
With `issue_reproducible` generated through flow or  Please package a standalone test case with https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docs/source/using_or_issue.md

vijayank88, 2023-07-17:
@maliberty Is this still relevant?

maliberty, 2023-07-17:
This seems done based on "Nothing further to resolve in this issue. ".  The other comments seem to be separate issues that should be filed as such.