The-OpenROAD-Project/OpenROAD-flow-scripts #174: 'no ap for <pin>' error during Detailed Routing step

lakshmi-sathi, 2021-10-01:
**Describe the bug**
The flow fails for these designs - a serv processor design and a tinyRocket processor design ([tinyRocket & serv_fr](https://github.com/bsg-idea/bsg_sky130_designs/tree/test_sky130/designs)) with an error 'no ap for w_mask_in[0] terminate called after throwing an instance of 'std::runtime_error' ' during the detailed routing step. They are using fakerams generated using [bsg_fakeram generator](https://github.com/bespoke-silicon-group/bsg_fakeram) as placeholders for its memories. 
The logs of the run are available here: 
https://github.com/bsg-idea/bsg_sky130_designs/tree/test_sky130/designs/serv_fr/logs
https://github.com/bsg-idea/bsg_sky130_designs/tree/test_sky130/designs/tinyRocket/logs

**Expected behavior**
Detailed Routing succeeding without 'no ap for ...' error

**Environment **
 - OS: Ubuntu 20.04
 - OpenROAD-flow v2.0-880-gd1c7001ad

**File Uploads**
Just have to place these folders in the 'designs/sky130hd/' directory and run it as usual for reproducing the issue: 
https://github.com/bsg-idea/bsg_sky130_designs/tree/test_sky130/designs/tinyRocket
https://github.com/bsg-idea/bsg_sky130_designs/tree/test_sky130/designs/serv_fr

**Additional context**
The congestion issue does not seem to appear on designs without SRAM macros. Different pin widths, pin spacings, and sizes were tried (both large and small) for the SRAM macros but still, the issue persists.

@tspyrou  @taylor-bsg

Stephanommg, 2021-10-04:
@lakshmi-sathi I ran tinyRocket with master branch of OpenROAD-flow-scripts and master branch of OpenROAD and it didnt get past macro place.

lakshmi-sathi, 2021-10-04:
What was the error faced? 
For me, if you see my logs, it completed Global Route: https://github.com/bsg-idea/bsg_sky130_designs/blob/test_sky130/designs/tinyRocket/logs/5_1_fastroute.log ,
and in detailed routing step it fails. 
Could you get some clue as to what could be causing the flow to fail on it?

Stephanommg, 2021-10-04:
@lakshmi-sathi first there were some warnings:

[INFO ODB-0226] Finished LEF file:  ./platforms/sky130hd/lef/sky130_fd_sc_hd_merged.lef
[INFO ODB-0222] Reading LEF file: ./designs/sky130hd/tinyRocket/sram_1024x32.lef
[WARNING ODB-0176] error: undefined layer (OVERLAP) referenced
[INFO ODB-0225]     Created 1 library cells
[INFO ODB-0226] Finished LEF file:  ./designs/sky130hd/tinyRocket/sram_1024x32.lef
[INFO ODB-0222] Reading LEF file: ./designs/sky130hd/tinyRocket/sram_64x32.lef
[WARNING ODB-0176] error: undefined layer (OVERLAP) referenced
[INFO ODB-0225]     Created 1 library cells

then, later :

[WARNING PDN-0042] No via added at (12413.761 12108.4 12417.361 12109.601) because the full height of met5 (1.6) is not covered by the overlap.
[WARNING PDN-0042] No via added at (12458.561 12108.4 12462.161 12109.601) because the full height of met5 (1.6) is not covered by the overlap.
[INFO PDN-0034]   - grid CORE_macro_grid_1 for instance frontend.icache.data_arrays_0.data_arrays_0_0_ext.mem
[WARNING PDN-0042] No via added at (218.64 12108.4 222.24 12109.601) because the full height of met5 (1.6) is not covered by the overlap.
[WARNING PDN-0042] No via added at (263.44 12108.4 267.04 12109.601) because the full height of met5 (1.6) is not covered by the overlap.
[INFO PDN-0015] Writing to database.
Command terminated by signal 9
Elapsed time: 10:55.81[h:]min:sec. Average CPU: 43%. Peak memory: 22312256KB.
make: *** [Makefile:431: results/sky130hd/tinyRocket/base/2_6_floorplan_pdn.def] Error 137

I was expecting you to run again with updated master branch on both (OR and ORFS) to make sure we are running the same version.

Stephanommg, 2021-10-05:
@lakshmi-sathi you can also do 

make detail_route_issue

to pack your synthesis result prior to detailed routing so that I can start detailed routing (for debugging) from the same point of you.

maliberty, 2021-10-06:
This comment makes little sense:

latest master of OR and 'd1c7001' of ORFS (the default one), the design goes through till routing.

I guess you meant

latest master of ORFS and 'd1c7001' of OR (the default one), the design goes through till routing.

Please clarify as d1c7001 is an OR commit.

lakshmi-sathi, 2021-10-07:
@maliberty  Sorry about the confusion, I meant "latest master of ORFS and 'd1c7001' of OR"

Thanks for the pointer!
I have collected the results using 'make detail_route_issue' (on latest master of ORFS and 'd1c7001' of OR'):
[tinyRocket](https://drive.google.com/file/d/1ecu0hiZ9z1J89I1idD6pBEHqi-YKdNWS/view?usp=sharing)
[Serv ](https://drive.google.com/file/d/1iPV30ZZ1wUQF1Evv9CchytA6PvI7s-Os/view?usp=sharing)

An update on what else I have tried:
I tried a core area of 11400x10920um and then tinyRocket is able to complete the flow. However, upon viewing the result most of the space is seen empty, like with Serv, and going for a core area less than the one mentioned, causes the 'Routing congestion too high' issue to pop up during global routiing. 
(The packed result above, is of this latest run)

Stephanommg, 2021-10-07:
@maliberty I think you missed a (at least part of) [comment ](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/174#issuecomment-936805575) from @lakshmi-sathi. He said that the no ap error is gone, but a "routing congestion too high" appeared. So the error is not the aps anymore. That is part of why I called you.

lakshmi-sathi, 2021-10-12:
@maliberty Thanks for that pointer!

I tried increasing MACRO_PLACE_HALO but that strangely seemed to result in a worse overflow during the global routing step. I tried increasing MACRO_PLACE_CHANNEL instead and that reduced the overflow and the Serv design completes the flow. 

The max(2*halo, channel) will be the spacing between the macros and between the macros and the boundary am I right? I was trying to understand why increasing the MACRO_PLACE_HALO could have worsened the overflow.
Is it as simple as increasing the halo or increasing the channel to increase the spacing between the macros (and with the boundary) so that there is more room to route the standard cells?

taylor-bsg, 2021-10-15:
@maliberty any thoughts on Lakshmi's questions?

lakshmi-sathi, 2021-11-27:
> I tried increasing MACRO_PLACE_HALO but that strangely seemed to result in a worse overflow during the global routing step. I tried increasing MACRO_PLACE_CHANNEL instead and that reduced the overflow and the Serv design completes the flow.

Does this make sense in the context that this issue was happening due to the `set_macro_extension` option?

maliberty, 2021-11-29:
@eder-matheus any thoughts on the above question?

eder-matheus, 2021-11-29:
@lakshmi-sathi @maliberty A larger MACRO_PLACE_CHANNEL could result in gcells with more resources near the macro pins, so it makes sense the reduced overflow in this case. I'm not so sure about the MACRO_PLACE_HALO though.

lakshmi-sathi, 2021-12-29:
@maliberty @eder-matheus Thanks for the reply! It helps.

As the original error is fixed, I am closing this issue.

closed 2021-12-29 by lakshmi-sathi