The-OpenROAD-Project/OpenROAD-flow-scripts #175: SRAM macro not getting picked up by the flow

lakshmi-sathi, 2021-10-01:
**Describe the bug**
The flow just doesn't seem to pick the SRAM macro for this manycore tile compute mesh design ([bsg_manycore_tile_compute_mesh_real](https://github.com/bsg-idea/bsg_sky130_designs/tree/test_sky130/designs)). It is hacking up an existing SRAM macro for the SRAM configurations that it requires. In the log for the macro placement step, this message is noted - 'No macros found: Skipping macro_placement'.
The logs of the run are available here: https://github.com/bsg-idea/bsg_sky130_designs/tree/test_sky130/designs/bsg_manycore_tile_compute_mesh_real/logs

**Expected behavior**
The SRAM macro getting picked up by the flow.

**Environment **
 - OS: Ubuntu 20.04
 - OpenROAD-flow v2.0-880-gd1c7001ad

**File Uploads**
Just have to place this folder in the 'designs/sky130hd/' directory and run it as usual for reproducing the issue: https://github.com/bsg-idea/bsg_sky130_designs/tree/test_sky130/designs/bsg_manycore_tile_compute_mesh_real

@tspyrou  @taylor-bsg

maliberty, 2021-10-01:
@rovinski do you have time to take a look at this?

tspyrou, 2021-10-01:
@rovinski assign to me if you don't have time to look at it.

rovinski, 2021-10-15:
Hi @lakshmi-sathi this just got lost in my notifications. It looks like you aren't reading in the memory lib. You need to add the line:
```
ADDITIONAL_LIBS = ./designs/$(PLATFORM)/$(DESIGN_NICKNAME)/sky130_sram_1kbyte_1rw1r_8x1024_8.lib
```
Blackbox verilog files have not been used for quite a while; OpenROAD requires a lib file for all macros, so you should remove the `VERILOG_FILES_BLACKBOX` line.

If this is a real RAM, you will need to include the GDS file under `ADDITIONAL_GDS` as well, otherwise you will not get a valid GDS (check for errors in `6_1_merge.log`)

lakshmi-sathi, 2021-10-18:
@rovinski I had previously tried with ADDITIONAL_LIBS but it wasn't working for me. I tried again just like you mentioned and here is what I get in the macro placement log:
```
OpenROAD 1 v2.0-880-gd1c7001ad
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: ./platforms/sky130hd/lef/sky130_fd_sc_hd.tlef
[INFO ODB-0223]     Created 11 technology layers
[INFO ODB-0224]     Created 25 technology vias
[INFO ODB-0226] Finished LEF file:  ./platforms/sky130hd/lef/sky130_fd_sc_hd.tlef
[INFO ODB-0222] Reading LEF file: ./platforms/sky130hd/lef/sky130_fd_sc_hd_merged.lef
[INFO ODB-0225]     Created 437 library cells
[INFO ODB-0226] Finished LEF file:  ./platforms/sky130hd/lef/sky130_fd_sc_hd_merged.lef
[INFO ODB-0222] Reading LEF file: ./designs/sky130hd/bsg_manycore_tile_compute_mesh_real/sky130_sram_1kbyte_1rw1r_8x1024_8.lef
[WARNING ODB-0205] The LEF UNITS DATABASE MICRON convert factor (2000) is greater than the database units per micron (1000) of the current technology.
[INFO ODB-0225]     Created 1 library cells
[INFO ODB-0226] Finished LEF file:  ./designs/sky130hd/bsg_manycore_tile_compute_mesh_real/sky130_sram_1kbyte_1rw1r_8x1024_8.lef
[INFO ODB-0127] Reading DEF file: ./results/sky130hd/bsg_manycore_tile_compute_mesh_real/base/2_3_floorplan_tdms.def
[INFO ODB-0128] Design: bsg_manycore_tile_compute_mesh
[INFO ODB-0130]     Created 1263 pins.
[INFO ODB-0131]     Created 35764 components and 206260 component-terminals.
[INFO ODB-0133]     Created 38105 nets and 134730 connections.
[INFO ODB-0134] Finished DEF file: ./results/sky130hd/bsg_manycore_tile_compute_mesh_real/base/2_3_floorplan_tdms.def
No macros found: Skipping macro_placement
Elapsed time: 0:05.22[h:]min:sec. Average CPU: 61%. Peak memory: 138704KB.
```

It still seems to skip the macro placement.
Here is the config I am giving (I have removed `VERILOG_FILES_BLACKBOX` and I have added `ADDITIONAL_LIBS`, I am yet to add the GDS, just thought to see if it'll pick up the macro first):
```
export DESIGN_NICKNAME = bsg_manycore_tile_compute_mesh_real
export DESIGN_NAME = bsg_manycore_tile_compute_mesh
export PLATFORM    = sky130hd

export VERILOG_FILES = ./designs/$(PLATFORM)/$(DESIGN_NICKNAME)/bsg_manycore_tile_compute_mesh.v \
        ./designs/$(PLATFORM)/$(DESIGN_NICKNAME)/hard_mem_1rw_byte_mask_d1024_w32_wrapper.v \
        ./designs/$(PLATFORM)/$(DESIGN_NICKNAME)/hard_mem_1rw_d1024_w46_wrapper.v

export SDC_FILE      = ./designs/$(PLATFORM)/$(DESIGN_NICKNAME)/bsg_manycore_tile_compute_mesh.sdc
export RCX_RULES     = $(PLATFORM_DIR)/rcx_patterns.rules

export ADDITIONAL_LEFS = ./designs/$(PLATFORM)/$(DESIGN_NICKNAME)/sky130_sram_1kbyte_1rw1r_8x1024_8.lef
export ADDITIONAL_LIBS = ./designs/$(PLATFORM)/$(DESIGN_NICKNAME)/sky130_sram_1kbyte_1rw1r_8x1024_8.lib

export _VDD_NET_NAME = vccd1
export _GND_NET_NAME = vssd1

export MACRO_PLACE_CHANNEL = 160 160

export DIE_AREA    = 0 0 3800 3080
export CORE_AREA   = 57 28 3420 2800

```

The issue still persists after these changes to the config. Could you please give me a hint of what else it could be that I am missing?

maliberty, 2021-10-18:
This is an issue for efabless to fix.  Please open an issue with that repo.  I'll keep this issue just for the warning vs error update.

taylor-bsg, 2021-10-20:
@mguthaus there seems to be a problem with interfacing the Sky130 SRAM LEFs to OpenRoad?

mguthaus, 2021-10-20:
HI @maliberty, where is that specified? I've been using 2000 in multiple tape outs through OpenLane.

maliberty, 2021-10-20:
@mguthaus https://github.com/google/skywater-pdk-libs-sky130_fd_sc_hd/blob/ac7fb61f06e6470b94e8afdf7c25268f62fbd7b1/tech/sky130_fd_sc_hd.tlef#L26

mguthaus, 2021-10-20:
I went through the OpenLane flow and my design passed DRC, LVS, and other checks, so *shrug*

taylor-bsg, 2021-10-20:
> Yes, it is fixed.

Thanks Matt!