The-OpenROAD-Project/OpenROAD-flow-scripts #207: Replace use of shell environment Variables with TCL Variables

tspyrou, 2021-11-02:
From @jjcherry56 
From  Clean up ORFS 

use tcl variable files instead of environment variablesglobal default fileplatform default filedesign values file

This allows us to remove all of the cruft that looks like this:if [file existsif [info exists

at the same time, remove all unused/obsolete variables that have collectedin the config.mk files. There are a lot.

Tasks 
In each TCL script we want to remove all of the calls to ::env.

In each TCL script we want to remove all calls to info and file exists since the tcl variable should tell us what to do.

Remove setting of unused variables many of which used to be needed but no longer are.

tspyrou, 2021-11-02:
@dralabeing @jjcherry56 fyi

maliberty, 2021-11-02:
Will openroad be launching yosys & klayout since all the control variables are in tcl?

How will a single flow step be run?

rovinski, 2021-11-02:
Tcl variables vs. environment variables has nothing to do with `if [info exists $xx]`. If the variable doesn't exist in Tcl, then you still have a problem.

I think there is a difference in expectations about how the flow should behave. @jjcherry56 has set up the OR repo so that the tests run in a single, clean step and the platforms are controlled to have all of the information needed.

This is not at all the working model I would use as an end-user if I were bringing up a platform. I would want to put the minimum required information to get a working minimum viable product. If the flow _can_ work without things like parasitics info, or a floorplan file, etc., then it should. I would find it incredibly frustrating to have to create every single file and set every variable before the flow could work.

Can the scripts use some cleanup? Sure. But that's orthogonal to Make vs. Tcl. I think a bigger problem is there need to be more thorough reviews for style and consistency. ASAP7 and sky130 IMO are a mess because they lack consistent structure. I see PRs get approved and merged which duplicate functionality or aren't coded consistently with similar features.

maliberty, 2021-11-02:
@rovinski you are welcome to be more involved in code reviews

This is more of a laundry list than a plan.

jjcherry56, 2021-11-02:
There is nothing in the spec I proposed that says make goes away @rovinski or the steps are merged @maliberty. I wouild love to get rid of the separate step files and makefile nonsense but that is not what this is proposing. Read it again. It merely cleans up the incredible mess that optional variables currently create. Variables should be optional. Either there is a default value set in a global set of values (tcl variables, since those are what drive openroad scripts) or in a platform file, or in a design file. Right now I get an issue and run global_route.tcl that has this:
if [file exists $env(PLATFORM_DIR)/setRC.tcl] {
  source $env(PLATFORM_DIR)/setRC.tcl
}

and the variable points to a non-existent file because the issue target is broken (many of them are) and it runs along
like there is not problem. There is a problem There are no layer RC values. It should no be optional and this style of
writing the scripts that allows them to be optional creates problems. 

This IS part of cleaning up the scripts. The days of 10 tools with command line args that the envars were designed
to contend with ended a long long time ago and it is time for the flow scripts to go up to reflect that and use tcl
variables so they look like normal tcl scripts instead copying idioms from all the people that didn't know even basic
tcl programming. It is embarrassing how poorly maintained they are and I believe this is an important first step in
cleaning them up.

Putting the minimum effort in bringing up a platforms means we get medhi copying values from a 12nm process into
a 130nm process description and then wondering why the timing in PT has so many buffers in it. Minimal effort means
we all pay for the laziness to debug all of the problems that are introduced by tolerance of the laziness. So, no, I 
completely disagree with "minimal effort". It has to be complete or there is no point in having it.

maliberty, 2021-11-02:
My problem is that many things are proposed here in one.  You responded to 'file exists' but ignored my questions entirely.  

I agree we should require setRC.tcl to be present as the flow is not very useful without it.  We only have derate.tcl for one pdk (idk if that is good or bad).

I see you propose to "remove all of the calls to ::env." without explaining how the flow will work without them.

Please drop the personal attacks.

rovinski, 2021-11-03:
> @rovinski you are welcome to be more involved in code reviews

I want to be, but I see PRs are often approved and merged before I even get a chance to see them. If I get requested, then I can usually get a review within a day. I haven't been touching ASAP7 nor sky130 PRs because the structure is still fundamentally broken.

@jjcherry56 I don't think you are thinking about this from the perspective of a community user. The point of OR is to actually make chips. Aside from sky130, we can't provide actual kits, so users need to set up their own platforms. **IF** we had the tools to automatically create or at least assist with the platform artifacts needed to run ORFS, that would be a different story because then it doesn't require work from the user. To date, I don't know anyone who has created a setRC.tcl that isn't directly affiliated with this project. There is no working flow to create it. We have been bootstrapping it from commercial tools or correlating it to SPEF data. How is a user supposed to run the flow to bootstrap parasitics when they can't run the flow without parasitics? Bogus data could be plugged, but I fail to see how that is more user-friendly than allowing the flow to continue without it. The same goes for fastroute.tcl, etc.

When someone runs GCC to compile their first C++ program, they don't want to worry about 10 million experimental flags or LTO or PGO or enabling the correct hardware optimizations. They want the minimum required to get something working. By forcing all of these files and variables to exist, we'd be forcing end users to make a complete platform before even letting them run a "hello world" for it.

Initial user experience makes a huge impact on whether that user will be retained or not. If we don't actually make this software friendly to users, then we're just designing the software to use ourselves. And then this project will lose support and die.

Separately, I don't understand how we are supposed to move all environment variables to tcl. We need the LEFs and LIBs defined to allow Make to create the wrapped / modified files to use in the flow. We need all sorts of other variables defined in the Makefile for proper flow control. Are we going to redefine those variables again in the tcl file?

Yes, I agree that the scripts could use a good cleaning and consistency pass. But I think that the proposed solutions aren't fully taking into account what's being sacrificed and/or are miscalculating the scope of the changes.

tspyrou, 2021-11-03:
This conversation is not converging, we will need to have some live discussions.

closed 2021-11-03 by tspyrou