The-OpenROAD-Project/OpenROAD-flow-scripts #216: Cleanup platform.mk for asap7

dralabeing, 2021-11-08:
Change platform specific configuration once and then design specific additions or changes that may be required.

vijayank88, 2021-11-09:
@dralabeing https://docs.google.com/spreadsheets/d/1u9JCNgLUV6LCRtO-gCgRIooH1K5IbUACjhXtZRW5Zlc/edit#gid=1778776042 refer Platform_Existing_config sheet for my comments about variable defined and not used, incorrect path.
Updated in sheet Platform_Updated_config. Check Waiting for Approval tab with **"info need"** whether these files can be added in future mean we can leave those empty as of now. 

With this update am able to complete a design successfully with same results. Once changes reviewed, will test other built-in design as well.

oharboe, 2022-12-30:
@dralabeing Close? I think this is fixed by lots of updates to platforms/asap7/config.mk?

dralabeing, 2022-12-30:
Yes

closed 2022-12-31 by maliberty