The-OpenROAD-Project/OpenROAD-flow-scripts #224: clean up fakeram liberty/lef libraries

jjcherry56, 2021-11-14:
I do not know how the fakeram liberty files were generated but they have a number of significant errors that should be
corrected. It may be easier to track down the script that wrote them an rerun it with reasonable inputs than to fix the
existing libraries.

There are duplicate copies of the liberty and lef files for the nangate45 fakerams.
For example:
./designs/nangate45/black_parrot/fakeram45_64x7.lib
./designs/nangate45/black_parrot/fakeram45_64x7.lef
./designs/nangate45/bp_multi_top/fakeram45_64x7.lib
./designs/nangate45/bp_multi_top/fakeram45_64x7.lef
./designs/nangate45/bp_fe_top/fakeram45_64x7.lib
./designs/nangate45/bp_fe_top/fakeram45_64x7.lef

There is no reason to have duplicates. It only complicate fixing problems with them.
All of the the fakeram libraries should be moved to platform/nanagate45/lib|lef so they can be shared.

------------------
The rise/fall transitions are completely unrealistic:
            rise_transition(fakeram45_64x7_mem_out_slew_template) {
                index_1 ("0.000, 0.002");
                values ("0.05, 1.00")
            }
            fall_transition(fakeram45_64x7_mem_out_slew_template) {
                index_1 ("0.000, 0.002");
                values ("0.05, 1.00")
            }

This is equivalent to a 500kohm drive resistance, which makes it impossible to connect the ram to anything without 
getting a max slew violation.
They should be replaced by the rise/fall_transition section for the output of the nangate45 std cell library BUF_X1 cell.
This will also require copying the axis templates.

------------------

Input capacitances are zero, which is completely unrealistic.
Again, copy the values from the BUF_X1 input.

    pin(clk)   {
        capacitance : 0.000;

    pin(we_in){
        capacitance : 0.000;

    pin(ce_in){
        capacitance : 0.000;

    bus(addr_in)   {
        capacitance : 0.000;

    bus(wd_in)   {
        capacitance : 0.000;

    bus(w_mask_in)   {
        capacitance : 0.000;

------------------

The capacitance units in fakerams are pf but the nangate45 library std cells are ff.
The fakerams should be using the same units.

maliberty, 2021-11-14:
@rovinski do you know how they fakeram libs were generated?  It looks like Tutu checked them in long ago.

rovinski, 2021-11-19:
This was discussed in the Monday meeting, but I'll add my comments here for visibility.

The fake RAMs were generated from [bsg_fakeram](https://github.com/bespoke-silicon-group/bsg_fakeram). It is a very generic generator that was originally designed (IIRC) in the course of 3 days as a stopgap to fill the need of not having memory generators for PDKs but needing them for EDA development. It has been updated a bit since then to fix some of the most glaring issues, but is not actively developed.

As mentioned in the meeting, bsg_fakeram is a good effort, but not a complete effort, to create a viable cross-platform memory generator. Discussion with others familiar with the matter indicated that would take a few months of dedicated work for a "complete" generator.

[OpenRAM](https://github.com/VLSIDA/OpenRAM) is a solution that creates real RAMs and should have viable characteristics because they have actual extracted parasitics. The problem is that 1) The generated RAMs (to date) tend to have much worse area than state-of-the-art equivalents. They also require a handful of specially-crafted cells (bit cell, sense amp, etc.). This would be fine for nangate45 and sky130 which already have some, but I don't believe ASAP7 has any. On the other hand, it looks like bsg_fakeram can't handle ASAP7 either (https://github.com/bespoke-silicon-group/bsg_fakeram/issues/10).

After looking at some of the fakeram code, it looks like the input cap problem could be fixed upstream. The transition values are unlikely to be fixed upstream without significant modelling effort.

jjcherry56, 2021-11-20:
The generators are just a bunch of print statements so it is trivial to change the liberty files to use a more reasonable output transition times. I don't see the json used to drive the bsg_fakeram generators anywhere in the ORFS hierarachy. Did they get saved someplace outside the repo? If we can't find them it shouldn't be that hard to reverse engineer them from the file names (and save them this time around).

jjcherry56, 2021-11-20:
I found the nangate45 json file [here](https://github.com/bespoke-silicon-group/bsg_fakeram/blob/master/example_cfgs/freepdk45.cfg).

vijayank88, 2021-11-25:
@jjcherry56 @maliberty @rovinski 
in bsg fake ram generator the default value in scripts/utils/cacti_config.py 
capacitance and dynamic power set in class_memory.py. Also units are wrong as per FreePDK45 technology.
```
-design objective (weight delay, dynamic power, leakage power, cycle time, area) 0:0:0:100:0
-deviate (delay, dynamic power, leakage power, cycle time, area) 20:100000:100000:100000:100000
-NUCAdesign objective (weight delay, dynamic power, leakage power, cycle time, area) 100:100:0:0:100
-NUCAdeviate (delay, dynamic power, leakage power, cycle time, area) 10:10000:10000:10000:10000
```
based above look up table fake ram is generated. So we need right values for below variables to come up with good library setup.
```
self.cap_input_pf = 0.005
self.t_setup_ns = 0.050
self.t_hold_ns  = 0.050
-design objective (weight delay, dynamic power, leakage power, cycle time, area) 0:0:0:100:0
-deviate (delay, dynamic power, leakage power, cycle time, area) 20:100000:100000:100000:100000
-NUCAdesign objective (weight delay, dynamic power, leakage power, cycle time, area) 100:100:0:0:100
-NUCAdeviate (delay, dynamic power, leakage power, cycle time, area) 10:10000:10000:10000:10000
```
Based on above variable attached json data created. If we pass right library data, maybe this issue will get fix.
[cacti.cfg.csv](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/7602783/cacti.cfg.csv)

maliberty, 2021-12-08:
I don't know what the next step is here and who is responsible for it.  @vijayank88 last comment seems open ended.

jjcherry56, 2021-12-08:
OR PR 1382 where I fixed them for the OR flow tests using a forked version of the generator https://github.com/jjcherry56/bsg_fakeram. Unfortunately the results are 7x the area of the ones we have been using and I am still trying to figure out why before committing them there or here because all hell broke lose in the OR regressions when I changed them.

vijayank88, 2021-12-09:
@jjcherry56 
May I get access to your fakeram fork ? Currently I'm not seeing any updates in https://github.com/jjcherry56/bsg_fakeram.
Possible I will try to figure out with scripts/look up table which is affecting area parameter.
Thanks

vijayank88, 2021-12-09:
@jjcherry56 
I just cloned your repo and checked. You made changes in library unit, not changed any values wrt., design look up table values.(I've done same work by changing units in script not helped after seeing above conversation)
I have checked with @tspyrou if library have mix of units STA will handle it.
As we need proper library values like power, rise/fall transition, capacitance and area in scripts/utils/cacti_config.py.
This will reduce required area and OR regression will pass(Hope more changes not required in regression)

tspyrou, 2021-12-09:
@vijayank88 this issue has already been resolved and the files updated. There is no need to look at it further.

maliberty, 2022-01-09:
Before this issue is closed could we document how the fakerams were generated for future reference (eg asap7)?

dralabeing, 2022-01-09:
Yes @maliberty will do once @vijayank88 wraps up testing with the updates made

closed 2022-03-27 by jjcherry56