The-OpenROAD-Project/OpenROAD-flow-scripts #230: make issue issues

jjcherry56, 2021-11-18:
The make issue target fails to include setRC.tcl

source $::env(PLATFORM_DIR)/setRC.tcl
[ERROR STA-0511] cannot open './platforms/asap7/setRC.tcl'.
Error: cts.tcl, 30 STA-0511

The TIME_CMD variable value is not portable and generates a syntax error
Error: vars-jpeg-asap7-base.tcl, 6 invalid command name "h:"

set env(TIME_CMD) "/usr/bin/time -f 'Elapsed time: %E[h:]min:sec. CPU time: user %U sys %S (%P). Peak memory: %MKB.'"

vijayank88, 2021-11-19:
@jjcherry56 
set below variable in design config.mk
export SET_RC_TCL              = $(PLATFORM_DIR)/setRC.tcl

This update got fixed in PR #210. Use this as temporary workaround.

rovinski, 2021-11-19:
@vijayank88 that's not the problem. This is completely separate. `make *_issue` is supposed to package a complete test case and it is missing files required to make the testcase run.

vvbandeira, 2021-12-07:
@jjcherry56 can you check the tar files generated by the #249 to see if it fixes the issue?
You can find them here: https://jenkins.openroad.tools/blue/organizations/jenkins/OpenROAD-flow-script-Public%2Fpublic_tests_all-pr/detail/PR-249-head/1/artifacts

closed 2021-12-13 by vvbandeira