The-OpenROAD-Project/OpenROAD-flow-scripts #231: Standardize on default CORNER variable definition for platforms

dralabeing, 2021-11-18:
Currently, there is no specific definition/selection for default CORNER in the platform config.mk. This is random for current PDKS including skywater130, nangate45, asap7 etc.

This should be clarified in the documentation as well.

dralabeing, 2021-11-18:
@maliberty Please re-assign to the appropriate team member if needed

maliberty, 2021-11-18:
@tspyrou @louiic @abk-openroad @rovinski  please comment on what you think the default corner should be.

We are currently inconsistent on our default corner, e.g.:
gf12: BC
tsmc65lp: WC

rovinski, 2021-11-18:
The reasoning for gf12 BC was that BC + margin = best chance of getting functional chips due to being extra cautious about hold slack.

Is multi-corner optimization in OR yet? Is it possible that the answer should be "both" and we add multi-corner support?

maliberty, 2021-11-18:
@rovinski if we had multi-corner I wouldn't have to ask....

rovinski, 2021-11-19:
Just checking ¯\_(ツ)_/¯
I would stick with BC.

vijayank88, 2021-11-19:
@maliberty For ASAP7 we will stick to BC. I'm facing issue with TC for DFF library read.

tspyrou, 2021-11-19:
The OpenROAD app itself supports multi corner optimization in the flow.
Unit tests like https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/test/gcd_sky130hd_fast_slow.tcl test it.
This feature needs to be propagated up to the flow scripts. The default can then be BC and WC with users able to add more corners as needed.

arlpetergadfort, 2021-11-23:
This is how we handle corners at the moment in our flow: https://github.com/The-OpenROAD-Project-private/OpenROAD-flow-scripts/tree/flow-corners I can open a PR if this is of interest to incorporate

tspyrou, 2021-11-23:
@arlpetergadfort please do make a PR and thanks for doing the work of integrating the new feature up to ORFS.

arlpetergadfort, 2021-11-23:
@tspyrou I opened #237

vijayank88, 2023-06-13:
@dralabeing 
Seems resolved. Can we close this?

closed 2023-06-13 by maliberty