The-OpenROAD-Project/OpenROAD-flow-scripts #233: error in running a synthesis with make

JayjeetAtGithub, 2021-11-18:
## Issue
I am trying to run a basic synthesis with by doing `make` from the command line and I am getting this error.

```
[INFO][FLOW] Using platform directory ./platforms/nangate45
(/usr/bin/time -f 'Elapsed time: %E[h:]min:sec. CPU time: user %U sys %S (%P). Peak memory: %MKB.' stdbuf -o L  -zz -rd design_name=RocketTile \
        -rd in_def=results/nangate45/tinyRocket/base/6_final.def \
        -rd in_files="./platforms/nangate45/gds/NangateOpenCellLibrary.gds  " \
        -rd config_file= \
        -rd seal_file="" \
        -rd out_file=results/nangate45/tinyRocket/base/6_1_merged.gds \
        -rd tech_file=./objects/nangate45/tinyRocket/base/klayout.lyt \
        -rm ./util/def2stream.py) 2>&1 | tee ./logs/nangate45/tinyRocket/base/6_1_merge.log
stdbuf: invalid option -- 'z'
Try 'stdbuf --help' for more information.
Command exited with non-zero status 125
Elapsed time: 0:00.00[h:]min:sec. CPU time: user 0.00 sys 0.00 (100%). Peak memory: 2088KB.
make: *** [Makefile:626: results/nangate45/tinyRocket/base/6_1_merged.gds] Error 125
```

Does anyone know how to fix this?

## Steps

```
./build_openroad.sh
./setup_env.sh

# uncomment a DESIGN_CONFIG in the ~/OpenROAD-flow-scripts/flow/Makefile

make
```

rovinski, 2021-11-19:
This is actually an error because KLayout is not found in your environment. The command is supposed to be the following:
https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/5cd66291c0dd8bbf0dd7e07b8550567bc65d7cb8/flow/Makefile#L625-L633

Where `$(KLAYOUT_CMD)` is set here:
https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/5cd66291c0dd8bbf0dd7e07b8550567bc65d7cb8/flow/Makefile#L241

If KLayout is not accessible from your shell, then `$(KLAYOUT_CMD)` becomes empty, and then `stdbuf` gets passed the arguments that are supposed to go to KLayout. I can see if I can try to insert an error for GDS merge if KLayout isn't detected.

JayjeetAtGithub, 2021-11-21:
Thanks a lot for the clarification.

closed 2021-11-21 by JayjeetAtGithub