The-OpenROAD-Project/OpenROAD-flow-scripts #235: (fake)rams for asap7

maliberty, 2021-11-22:
This is to follow up on the ram discussion in https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/210

Real or fake rams are needed for asap7.

dralabeing, 2021-11-29:
@vijayank88 Please update the test status with the changes proposed to the lib values.

vijayank88, 2021-12-13:
@dralabeing 
I've manually derived LIB/LEF able to complete flow, but 216 DRC violations left after detailed routing. But its should not be acceptable right for PR?

maliberty, 2021-12-13:
I would be good to understand why there are so many drc

vijayank88, 2021-12-13:
@maliberty  shall I share tar file for reference?
DRC violations due to library that I created manually

maliberty, 2021-12-13:
If you don't see a clear problem then do share it.

rovinski, 2021-12-13:
Does ASAP7 have unidirectional routing? I wonder if it's related to off-grid pins on the macros because gf12 macros are wrapped but these are not.

maliberty, 2021-12-13:
> Does ASAP7 have unidirectional routing? I wonder if it's related to off-grid pins on the macros because gf12 macros are wrapped but these are not.

Yes