The-OpenROAD-Project/OpenROAD-flow-scripts #248: [ERROR DRT-0073] No ap for wishbone.bd_ram.spram_32_256/din0[0]

vijayank88, 2021-12-07:
**Describe the bug**
During detail routing am facing below issue
```
[INFO DRT-0165] Start pin access.
[INFO DRT-0076]   Complete 100 pins.
[INFO DRT-0076]   Complete 200 pins.
[INFO DRT-0076]   Complete 300 pins.
[INFO DRT-0076]   Complete 400 pins.
[INFO DRT-0076]   Complete 500 pins.
[INFO DRT-0076]   Complete 600 pins.
[INFO DRT-0076]   Complete 700 pins.
[ERROR DRT-0073] No ap for wishbone.bd_ram.spram_32_256/din0[0].
terminate called after throwing an instance of 'std::runtime_error'
  what():  DRT-0073
Elapsed time: 0:14.68[h:]min:sec. CPU time: user 28.59 sys 1.07 (201%). Peak memory: 1544204KB.
make: *** [results/asap7/ethwish/base/5_route.def] Error 6
```
I'm trying to use spram_32_256_asap7.lef, spram_32_256_asap7_TT_1p0V_25C.lib files as hard macro for eth_spram_256x32. LEF/LIB files I derived from nangate45 technology, as ASAP7 not yet supported by SRAM generator.

**Expected behavior**
Successfully complete flow.

**Environment (please complete the following information):**
Please only report issues for supported OSs.
 - OS: Ubuntu 20.04
 - OpenROAD-flow commit : v2.0-1978-gc1af9a878

**File Uploads**
Attached detail_route tar file detail_route_ethwish_asap7_base_2021-12-07_06-38
Size is more then 45MB. So given google drive access path

maliberty, 2021-12-07:
When I run your test case I get:
[ERROR ORD-0001] ./designs/asap7/ethwish/spram_32_256_asap7.lef does not exist.

it appears the packaging missed at least one file.

vijayank88, 2021-12-08:
> When I run your test case I get: [ERROR ORD-0001] ./designs/asap7/ethwish/spram_32_256_asap7.lef does not exist.
> 
> it appears the packaging missed at least one file.

@maliberty  shared in google drive missing LEF file

maliberty, 2021-12-08:
Your ram is in the wrong orientation.  M4 pins on the south which is an H layer and M3 pins on the west which is a V layer.

maliberty, 2021-12-08:
" I derived from nangate45 technology, as ASAP7 not yet supported by SRAM generator." is your problem as the layer stack doesn't align between these processes.

vijayank88, 2021-12-08:
@maliberty is manually derived lib and lef won't be supported?

maliberty, 2021-12-08:
@vijayank88  It isn't acceptable if it makes no sense.   You need the pins on the right layers.  The tool doesn't care how it was derived.  Since this is fake data you can just swap the pin layers to be correct for asap7.

QuantamHD, 2021-12-08:
@maliberty is there something we could do to improve the error message for pin access. Not everyone will be familiar with terms like ap and DRT-0073. Making this more user friendly would help the community solve their issues more quickly.

maliberty, 2021-12-08:
@QuantamHD do you have a suggestion as a what we might say?

QuantamHD, 2021-12-08:
Instead of 
```
No ap for wishbone.bd_ram.spram_32_256/din0[0].
```

I would say this might be more clear

```
The detailed router couldn't find an access point to the pin named wishbone.bd_ram.spram_32_256/din0[0]. 

This could be caused by a few things:

1. The pin is block from all sides above, and below by other nets or vias.
2. The pin layers may not match the detailed routers pin layer configuration. See {insert_link_here} on how you can resolve this.
3. Other likely causes... 
```

maliberty, 2021-12-08:
I like it.  You are in charge of user messages :)

rovinski, 2021-12-09:
In commercial tools, they often leave a short error message in the log and provide a more detailed message if you use `help [error code]` or `man [error code]`. It would be great if we eventually had a similar structure.

So in this example the message could stay as is and then `help DRT-0073` could provide a message similar to what @QuantamHD wrote.

vijayank88, 2021-12-09:
@rovinski  I agree with your suggestion.
This is also part of documentation work which already discussed with @dralabeing. 
This should be collaborative effort from designer message and Developer Error code

vijayank88, 2021-12-09:
> @vijayank88 It isn't acceptable if it makes no sense. You need the pins on the right layers. The tool doesn't care how it was derived. Since this is fake data you can just swap the pin layers to be correct for asap7.

@maliberty Even numbers are Horizontal layer, Odd number are vertical layers in ASAP7. 
But in sky130 and nangate45 its inverse. Updated my configuration and its got fixed.

QuantamHD, 2021-12-09:
@rovinski 

I disagree that it should be a separate tool for a couple of reasons.

1. We don't want to break the flow of our users. Asking them to run a separate command breaks their flow, and may lead to getting distracted, or forgetting what they were trying to do.
2. If messages exist separate from the codebase they're documenting they're likely to go out of date.

I so think we should adopt an internationalization library for string messages in the long term so that we can support users in multiple languages. That would mean we have separate files for strings that we could use to generate such man tools, and web pages.

mithro, 2021-12-09:
clang having developer friendly error messages (for things like C++ template errors) is one of the reasons people started switching away from GCC. I would *highly* recommend looking into them.

See https://clang.llvm.org/diagnostics.html https://gcc.gnu.org/wiki/ClangDiagnosticsComparison and https://blogs.gnome.org/mortenw/2014/01/27/gcc-vs-clang-for-error-messages/

David Malcolm at RedHat has some great blog posts about the improvements in GCC in the last couple of years in this area;
 - https://developers.redhat.com/blog/2018/03/15/gcc-8-usability-improvements
 - https://developers.redhat.com/blog/2019/03/08/usability-improvements-in-gcc-9
 - https://hub.packtpub.com/gcc-9-will-come-with-improved-diagnostics-simpler-c-errors-and-much-more/

Error message usability matters.

mithro, 2021-12-09:
> In commercial tools, they often leave a short error message in the log and provide a more detailed message if you use `help [error code]` or `man [error code]`. It would be great if we eventually [...]

It is *good* to have short error codes associated with error messages. 

It is **bad** to require a seperate tool to make those error codes useful for all the reasons @QuantamHD points out.

mithro, 2021-12-09:
FYI - Short error codes tend to be mostly useful for toolchain developers, not end users.

mithro, 2021-12-09:
I recommend also reading the "Not just for humans" section in the https://developers.redhat.com/blog/2019/03/08/usability-improvements-in-gcc-9 blog post.

mithro, 2021-12-09:
FYI - All open source projects should support localization (l10n) and internationalization (i18n). Please just use gettext so you can integrate into various translation tooling and efforts. See;
 - https://www.gnu.org/software/gettext/manual/gettext.html
 - https://en.wikipedia.org/wiki/Gettext

maliberty, 2021-12-09:
@rovinski when the logger's error() is called it always throws an exception (see https://github.com/The-OpenROAD-Project/OpenROAD/blob/33aaefeb2f9d9a80bfafec464c338040b9b65f17/src/utl/include/utl/Logger.h#L152)

So unless someone catches it, the execution will stop.  We do have code to propagate the exception from c++ to tcl in the swig wrappers.

maliberty, 2021-12-09:
In all the clang/gcc examples the error message itself is one line long and the remaining lines are annotated source code.  None are as verbose as what @QuantamHD proposes.

rovinski, 2021-12-09:
If true, then there could be detailed messages (such as https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/248#issuecomment-989315060) for errors and short messages with "Use help [warn code]" for warnings? I'd be okay with that.

The other issue that brings up is where to put the long messages. Putting it right in the code like we do now seems like a hassle and breaks code flow.

maliberty, 2021-12-09:
@mithro @QuantamHD Internationalization is a huge project that I don't see the resources for.  We haven't even fully completed the logger conversion after more than six months.

maliberty, 2021-12-09:
@vijayank88 is there anything further to do here on your original issue?

mithro, 2021-12-09:
gcc / llvm error messages have gotten a lot better but they are far from perfect.

Error messages (+ logging) are like documentation and testing in that there is **always** more work to do and they can **always** be improved. Like documentation and testing they are also frequently neglected and not enough effort is put into them.

If you want we can put together a set of principles and style guide with examples of how OpenROAD (and related tools) should aspire to achieve. (This would probably be done by steal liberally from big projects with established best practices.) These can start with a set of base principles like "a user should always have an actionable next step", should include guidance on when to use the various levels and then include more specifics about how they should be formatted.

mithro, 2021-12-09:
BTW EDA tools are **famously** bad at logging, warnings and error best practices. It is not uncommon for things that have no effect ending up as "critical warnings" and messages about critical design failures being "info messages". To make things worse they also have huge issues with excessive logging to the point that the log files associated with a design run start to compare to the output GDS files!

maliberty, 2021-12-09:
@mithro that's quite true but it is also true that users have vastly different expectations.  Some want to check every little detail and others don't care if the tool catches fire so long as they get a result.

QuantamHD, 2021-12-10:
Our goal isn't "a result", it's a brand new way to build things for an audience that's much larger than the existing market. We want people to be proud of the tools they use, and to be excited by what comes next rather than discouraged by cruft

@maliberty I think that's the difference between this project, and a commercial vendors.

maliberty, 2021-12-10:
@QuantamHD nobody is arguing for bad error messages.  I am just pointing out that among just the few people on this list we don't even have agreement as to what a good message is.  We currently have >3k messages and curating them all is a big job (in fact we just did a pass to fix grammar and especially bad ones).  

In the spirit of open source I encourage people to jump in and make things better rather than waiting for the small core team to prioritize message clarity over the numerous other competing interests.

QuantamHD, 2021-12-10:
Totally understood Matt. I don't expect you, or anyone on the core team to do this work. I'd be happy to help change these things myself. I think what this discussion has turned into though probably deserves a different title, and is really trying to get to the heart of what the best user experience is. I definitely get your and @rovinski's point about not inundating the logs with verbose logs.

To that end I think the discussion is useful to feel out the right mindset rather than embarking on a massive project.

vijayank88, 2021-12-10:
> @vijayank88 is there anything further to do here on your original issue?

@maliberty 
end up with below error
```
[INFO] Merging GDS/OAS files...
	./platforms/asap7/gds/asap7sc7p5t_27_R_1x_201211.gds
	./designs/asap7/ethwish/spram_32_256_asap7.gds
ERROR: ./util/def2stream.py:130: Invalid record or data type (position=128, record number=6, cell=spram_32_256_asap7) in Layout.read
  ./util/def2stream.py:130 (class exceptions.RuntimeError)
Elapsed time: 0:13.45[h:]min:sec. CPU time: user 12.84 sys 0.60 (99%). Peak memory: 1047624KB.
cp results/asap7/ethwish/base/6_1_merged.gds results/asap7/ethwish/base/6_final.gds
cp: cannot stat 'results/asap7/ethwish/base/6_1_merged.gds': No such file or directory
make: *** [results/asap7/ethwish/base/6_final.gds] Error 1
```

rovinski, 2021-12-10:
@vijayank88 where did you get the spram GDS from?

vijayank88, 2021-12-10:
@rovinski  I've used wrong GDS file for spram. Updated makefile and running again.

closed 2021-12-10 by vijayank88