The-OpenROAD-Project/OpenROAD-flow-scripts #285: All designs fail: Error: 2_floorplan.sdc, 1 invalid command name ������������µRˎÂ0"

yurivict, 2021-12-28:
Error:
```
[INFO ODB-0134] Finished DEF file: ./results/nangate45/aes/base/2_floorplan.def
Error: 2_floorplan.sdc, 1 invalid command name ������������µRˎÂ0"
```

Revision: 791e2920ad5edccfb0541cf0507e4e05f449adbf
OpenRoad revision: 719c6dc61

maliberty, 2021-12-28:
This problem was fixed earlier and passes in the current CI.  It is unclear what you are doing differently to trigger the issue.

yurivict, 2021-12-29:
> This problem was fixed earlier and passes in the current CI. It is unclear what you are doing differently to trigger the issue.

What commit fixed this problem? Because I am still seeing it with the current OpenRoad and ORFS revisions.

maliberty, 2021-12-29:
https://github.com/The-OpenROAD-Project/OpenROAD/commit/95ffbc5f8aefb66b197525f2098e4bed443e0199

yurivict, 2021-12-30:
For me 2_floorplan.sdc is written gzipped but it is read as cleartext - this causes the failure.

vijayank88, 2022-01-04:
@yurivict 
Is this issue still open?
If yes provide reproducible tar file with below command
```
make floorplan_issue
```

jjcherry56, 2022-04-04:
This issue was triggered by the addition of abc submodule that included the sources of a duplicate conflicting version of zlib. I fixed it by moving abc into its own namespace instead of the global namespace:
7e4667a65 James Cherry 2022-01-17 abc namespace

closed 2022-04-04 by jjcherry56