The-OpenROAD-Project/OpenROAD-flow-scripts #288: Tracks.tcl documentation request

QuantamHD, 2021-12-29:
https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/f801b4b8b321a6ea8ba4ffa84c939b6d1d8f658f/flow/platforms/asap7/openRoad/make_tracks.tcl#L16-L22

In the above snippet for ASAP7 there are multiple track definitions for M2. What is the purpose of these repeat definitions? I also got a question inside Google about why the tracks need to be defined manually. Can this information be picked up from the techlef?

maliberty, 2021-12-29:
@louiic is the expert on this so I hope he will comment.  LEF does not give you a track pattern directly.

maliberty, 2021-12-29:
I recall now that in asap7 the row height is 7.5 M2 tracks.  So if you used the pitch you would get irregular track patterns.   If you look at the pattern:
[image]
There are six useable tracks per row and they are spaced at min pitch inside the power stripes.

QuantamHD, 2021-12-29:
What control the track configuration for a given technology? Is it encoded by the standard cell layout indirectly?

maliberty, 2021-12-29:
The LEF has some syntax to express a track pattern (OFFSET and PITCH).  However in asap7 that is insufficient due to the above issue and so we have to manually define the pattern.  In other PDKs the LEF is sufficient and you will not see a MAKE_TRACKS setting in the platform config.mk.

In this case, which is common, the issue is the cell height and the power rails create the need for a particular pattern.  Other situations are possible so we leave the manual option open.

QuantamHD, 2021-12-29:
Could I write some thing that could auto detect a track pattern from a standard cell, or would that not be generally applicable?

maliberty, 2021-12-29:
I suspect there are more corner cases that @louiic could comment on.   However making something that can automate the known PDKs would be nice.

vijayank88, 2021-12-30:
@QuantamHD
https://github.com/The-OpenROAD-Project/OpenROAD/tree/fed16d7ea74d7fd82d08022b60513123ea354622/src/ifp#make-tracks : that's the quick info i found on documentation

vijayank88, 2023-07-17:
@QuantamHD Any update on this issue?