The-OpenROAD-Project/OpenROAD-flow-scripts #289: Detail Route Issue on BlackParrot design: 'no ap for <pin>'

lakshmi-sathi, 2021-12-29:
**Describe**
BlackParrot processor design not able to pass Detailed Routing.
Error Message: 
```
[ERROR DRT-0073] No ap for multi_top.rof1_0__core.fe.icache_1.metadata_mem.macro_mem.mem/w_mask_in[6].
terminate called after throwing an instance of 'std::runtime_error'
  what():  DRT-0073
Command terminated by signal 6
Elapsed time: 2:34.81[h:]min:sec. Average CPU: 114%. Peak memory: 8527904KB.
make: *** [Makefile:544: results/sky130hd/black_parrot/base/5_route.def] Error 134
```
Different MACRO_PLACE_HALO and MACRO_PLACE_CHANNEL have been tried (20 to 400 & 200 to 500 respectively), different place densities and areas have been tried, and also the congestion map indicates very low congestion but still running into the same detailed routing error.

**Expected behavior**
The design would pass detailed routing.

**Screenshots**
Attaching the congestion map:
[image]
The congestion map strangely seems to indicate a very low congestion?

**Environment:**
 - OS: Ubuntu 20.04
 - OpenROAD-flow-scripts commit: 791e2920ad5edccfb0541cf0507e4e05f449adbf
 - OpenROAD commit: fed16d7ea74d7fd82d08022b60513123ea354622

**File Uploads**
[detail_route_issue ](https://drive.google.com/file/d/1kwV_w6XG9Egp3gvQlthL1at8iU9WH5xP/view?usp=sharing)

**Additional context**
All other designs in PR #240 are able to complete the flow. Only BlackParrot faces this detailed routing issue, BlackParrot is also the largest design.
Previously, before the 'set_macro_extension' command was removed from the fastroute config., the design was failing at the Global Routing step with 'routing congestion too high error'. After the fix to that issue, the design now fails at Detailed Routing with 'No ap for multi_top.rof1_0__core.fe.icache_1.metadata_mem.macro_mem.mem/w_mask_in[6]'

taylor-bsg, 2021-12-31:
@tspyrou

dralabeing, 2022-01-03:
@vijayank88 Can you re-run this test to reproduce this issue and track the origin? this doesn't seem to be a problem with congestion-

maliberty, 2022-01-03:
Pin access issues don't relate to congestion so much as off grid or blocked pins.  Its more helpful to include a zoom in of the pin in question.  I'll take a quick look at the pin.

maliberty, 2022-01-03:
In your fakeram you have two pins on top of eachother:
  PIN w_mask_in[6]
    DIRECTION INPUT ;
    USE SIGNAL ;
    SHAPE ABUTMENT ;
    PORT
      LAYER met3 ;
      RECT 0.000 195.550 0.900 196.450 ;
    END
  END w_mask_in[6]
  PIN rd_out[0]
    DIRECTION OUTPUT ;
    USE SIGNAL ;
    SHAPE ABUTMENT ;
    PORT
      LAYER met3 ;
      RECT 0.000 195.550 0.900 196.450 ;
    END
  END rd_out[0]

closed 2022-01-03 by maliberty

lakshmi-sathi, 2022-01-03:
@maliberty Thanks for taking a look and pointing this out.
I had faced this pin overlap once and had taken steps to handle it, so I wasn't expecting it to come up again and hence hadn't thought in that direction.