The-OpenROAD-Project/OpenROAD-flow-scripts #290: Can't find instance if use ABC_CLOCK_PERIOD_IN_PS for sky130hd chameleon

vijayank88, 2021-12-29:
**Describe the bug**
* [Toolname] for runtime issues (e.g. Macro placement [mpl]])
For ORFS sky130hd chameleon design passes successfully all flow stages without 
```
export ABC_CLOCK_PERIOD_IN_PS = 1000
```
If we use above variables failed to complete placement stage

Here is my findings:

    Without ABC_CLOCK_PERIOD_IN_PS:
```
[INFO][FLOW-xxxx] Using manual macro placement file ./platforms/sky130hd/chameleon/chameleon.macro_placment.cfg
_c869220600000000_p_odb__dbBlock
1000
file7
LINE : _3252_ R0 200 150
LINE : _3253_ R0 1600 150
LINE : _3254_ R180 1289.845 3339.36
LINE : ahb_sys_0_uut.S0.CACHE R0 2100 2650
LINE : ibex_core R0  2150 1700
LINE : ahb_sys_0_uut.apb_sys_inst_0 R0 1470 2643
LINE : 
```
    With ABC_CLOCK_PERIOD_IN_PS:
```
[INFO ODB-0134] Finished DEF file: ./results/sky130hd/chameleon/base/2_3_floorplan_tdms.def
[INFO][FLOW-xxxx] Using manual macro placement file ./platforms/sky130hd/chameleon/chameleon.macro_placment.cfg
_389be40600000000_p_odb__dbBlock
1000
file7
LINE : _3252_ R0 200 150
Error: macro_place.tcl, 145 Cannot find instance _3252_
```
There was issue with ```ord::get_db_block```
Is there any fix to get right db_block?

**Expected behavior**
Should pass all stages of the ORFS flow.

rovinski, 2021-12-29:
@vijayank88 this is because the way chameleon is set up is very brittle. `_3252_` is an auto-generated instance name from yosys. The creator(s) of this design ran synthesis to get the auto-generated instance names of the macro(s), then put those names in the config file.

By changing `ABC_CLOCK_PERIOD_IN_PS`, you are causing a perturbation in yosys which causes the instance name of the macro to be changed to something else, thus `_3252_` doesn't exist. The best solution would be to find a workaround that doesn't require an auto-generated instance name in the config file.

maliberty, 2021-12-29:
Why is platforms/sky130hd/chameleon in the platform and not in the design?  Why does this design need a manual macro placement at all?

rovinski, 2021-12-29:
> Why does this design need a manual macro placement at all?

Same issue as in https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/217#issuecomment-964659932. Macro placer does not do a good job and leads to failures.

maliberty, 2021-12-29:
That PR is danglig with "@rovinski what remains to do here?"  The chameleon issue had been resolved.

vijayank88, 2023-06-13:
chameleon is part of CI. Shall we close this as resolved? @maliberty

closed 2023-06-13 by maliberty