The-OpenROAD-Project/OpenROAD-flow-scripts #298: GDS generated with unmapped layers in klayout for sky130hd

tuohangzeng, 2022-01-01:
**Describe the bug**

* [Flow] The flow script can successfully generate the results (including the gds) with all the routings. But when the gds is imported into klayout, the metal layers (e.g. met2 drawing) becomes empty with the [lyp](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/platforms/sky130hd/sky130hd.lyp) file loaded. Screenshots for klayout before and after loading the lyp file is shown below.

**Expected behavior**
* Routings are correctly mapped to the different metal drawing layers.

**Screenshots**
* Before loading lyp
[image]

* After loading lyp (with "Add other layer entries" and "Hide empty layers" selected) 
  - Several layers only have the "layer/datatype" name
[image]

**Environment (please complete the following information):**
 - OS: CentOS Linux 7
 - OpenROAD-flow commit 1a27533

**File Uploads**
[gcd_sky130hd_base_2022-01-01_17-44.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/7798054/gcd_sky130hd_base_2022-01-01_17-44.tar.gz)

rovinski, 2022-01-02:
The `.lyp` file doesn't change any properties of the GDS; it only changes the way it is rendered in the viewer (layers, colors, labels, etc.). That being said, it definitely looks like some layers aren't being mapped correctly by the `.lyt` file.

maliberty, 2022-01-02:
@rovinski yes and I can't reproduce the problem @ ucsd.  I am guessing an old version of klayout at Michigan but I don't know.  m2 is being mapped to 7/0 which is wrong and not in the lyt.

tuohangzeng, 2022-01-02:
Thank you, @rovinski and @maliberty! It was indeed related to the .lyt file. After doing a diff with OpenLane and making a few changes, all layer mappings are correct.

closed 2022-01-02 by tuohangzeng

msaligane, 2022-01-02:
@tuohangzeng Please remember to send a PR to update this file in orfs as well.

maliberty, 2022-01-02:
What did you change?  I don't see any errors when I run ORFS with the current file.