The-OpenROAD-Project/OpenROAD-flow-scripts #303: Power Strip connections (met4 met5) not correctly mapped in GDS

tuohangzeng, 2022-01-05:
**Describe the bug**
* [Flow] Power strips that were on met4 and met5 are mapped to layers cldntm.mask and rdlcon.drawing in GDS. There are also layers 5/0 and 236/0 that do not have mappings. The flow is running on the latest commit.

**Expected behavior**
The power strips should be mapped onto met4 and met5 respectfully. 

**Screenshots**
In OpenRoad -gui, the metals show up on met4 and met5:
[image]

In the GDS (opened in klayout and after loading the lyp), the layers are incorrectly mapped:
[image]

**Environment**
 - OS: CentOS7
 - OpenROAD-flow commit d4e6903

**File Uploads**
[orfs_sky130hd_gcd.zip](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/7818308/orfs_sky130hd_gcd.zip)

maliberty, 2022-01-05:
Will you check in https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/299 fixes this problem?

tuohangzeng, 2022-01-05:
Actually, this is likely a klayout bug. I just checked their release note and noticed that 0.27.6 fixed a bug in 0.27.5:

Bugfix: [#942](https://github.com/KLayout/klayout/issues/942) Technology file routing suffix string typo

I will update klayout and rerun the gds generation part of the flow and update my results here. Thank you very much!

tuohangzeng, 2022-01-06:
Yes, it was indeed a klayout issue. It is now working after updating klayout.

closed 2022-01-06 by tuohangzeng