The-OpenROAD-Project/OpenROAD-flow-scripts #305: asap7 flow: make seems to be broken

fabrizioferrandi, 2022-01-10:
Dears,
I'm experimenting with the single script run_all.tcl.
The flow can be activated by a simple `make all` for what I've understood.

In case asap7 design flows are considered,  `make all` stops after a while because merged.lib is missing.
To reproduce the issue, I've removed all the generated files from directories: results, logs, reports, and objects.
The issue could be easily fixed by adding to the all rule the `$(DONT_USE_SC_LIB) `:
so the following line in the flow Makefile 
`all: $(SDC_FILE) $(WRAPPED_LIBS) $(DONT_USE_LIBS) $(OBJECTS_DIR)/klayout.lyt $(WRAPPED_GDSOAS)
`
need to be changed in:
`all: $(SDC_FILE) $(WRAPPED_LIBS) $(DONT_USE_LIBS) $(DONT_USE_SC_LIB) $(OBJECTS_DIR)/klayout.lyt $(WRAPPED_GDSOAS)
`

Once you fix this issue, another issue comes out. It seems that `fillcell.tcl` script does not complete successfully with this error message:
```
Starting fill cell
[INFO DPL-0001] Placed 451916 filler instances.
[WARNING DPL-0005] Overlap check failed (451916).
[ERROR DPL-0033] detailed placement checks failed.
Error: fillcell.tcl, 21 DPL-0033
Command exited with non-zero status 1
```

The same fillcell.tcl step successfully completes when the default make command is used. I tried to investigate from where the issue came out, but I could not find any solution.

To reproduce the issue, just uncomment one of the asap7 designs. I did with `DESIGN_CONFIG=./designs/asap7/aes/config.mk`
and then give the command `make all`

maliberty, 2022-01-10:
Are you using the latest OR?  It passed in last night's CI.

fabrizioferrandi, 2022-01-10:
I'm using `OpenROAD v2.0-2568-ge6d5e2d7d`on an Ubuntu 20.04.

maliberty, 2022-01-10:
That build is from Dec 18 - can you try the current master?

fabrizioferrandi, 2022-01-11:
The problem is still there even with `OpenROAD v2.0-2631-gcd7a2f497`.

maliberty, 2022-01-12:
I'll fix this issue but in general the 'all' target is somewhat experimental and not our standard flow.  Is there a reason you need to use it?  I guess you'll find more issues.

maliberty, 2022-01-12:
https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/310

fabrizioferrandi, 2022-01-12:
Thank you very much for looking into the issue.
I'm trying to integrate OpenRoad into the High-level synthesis tool Bambu. It's easier for us to call a single bash script where openroad is called once with a script similar to run_all.tcl. Bambu interacts with the RTL synthesis backends through scripts generated starting from XML templates. Here you may see a draft example of the OpenRoad XML template I'm working on: [link](https://github.com/ferrandi/PandA-bambu/blob/panda-0.9.7-dev/etc/devices/Generic_devices/Generic-yosysOpenROAD.xml)

maliberty, 2022-01-12:
I fear your approach may be brittle as we make changes to ORFS.  Why not just call make (or wrap that in a bash script)?

maliberty, 2022-01-12:
#310 is merged so this particular problem is resolved though others may remain.

closed 2022-01-12 by maliberty