The-OpenROAD-Project/OpenROAD-flow-scripts #307: Unable to build with docker

AndreKuru, 2022-01-12:
**Describe the bug**
I'm trying to follow the steps mentioned in the [OpenROAD documentation](https://openroad.readthedocs.io/en/latest/user/BuildWithDocker.html) for Building from sources using docker option, but the script build_openroad.sh is failling.

**Expected behavior**
Finish build_openroad.sh without errors.

**Environment:**
 - OS: ArchLinux
 - OpenROAD-flow commit a24a0607146bfd217a1928e928c2e68cc37f2521

**Log from build_openroad.sh**
`[INFO FLW-0002] Updating git submodules.
Submodule path 'tools/OpenROAD': checked out 'fed16d7ea74d7fd82d08022b60513123ea354622'
Submodule path 'tools/OpenROAD/src/sta': checked out 'e0998d590caaec71fe103391cf8dc422ddccf412'
[INFO FLW-0000] Using docker build method. This will create a docker image tagged 'openroad/flow-scripts'.
[INFO FLW-0020] Building docker image for Yosys.
Using default tag: latest
latest: Pulling from openroad/yosys-dev
Digest: sha256:f1dc3288b3093ebcf7c5d4c48b286d0d2cc76fc8b79f046bdc37a5b53b1fbd4e
Status: Image is up to date for openroad/yosys-dev:latest
docker.io/openroad/yosys-dev:latest
Sending build context to Docker daemon  239.3MB

Step 1/4 : FROM openroad/yosys-dev AS builder
 ---> 2842cd8a1332
Step 2/4 : COPY . /yosys
 ---> 72d1dff8fd6f
Step 3/4 : WORKDIR /yosys
 ---> Running in 7521b3ac9237
Removing intermediate container 7521b3ac9237
 ---> bef54e7472fd
Step 4/4 : RUN make PREFIX=/install CONFIG=gcc ABCREV=bafd2a7 ABCURL=https://github.com/berkeley-abc/abc install -j$(nproc)
 ---> Running in cf6dc3c5e160
[  0%] Building kernel/version_UNKNOWN.cc
[  0%] Building kernel/driver.o
[  1%] Building kernel/register.o
[  1%] Building kernel/rtlil.o
[  1%] Building kernel/log.o
[  2%] Building kernel/calc.o
[  2%] Building kernel/yosys.o
[  3%] Building kernel/cellaigs.o
[  3%] Building kernel/celledges.o
[  3%] Building kernel/satgen.o
[  4%] Building kernel/mem.o
[  5%] Building libs/bigint/BigIntegerUtils.o
[  6%] Building libs/bigint/BigUnsignedInABase.o
[  6%] Building libs/sha1/sha1.o
[  6%] Building libs/json11/json11.o
[91mkernel/satgen.cc: In member function 'bool Yosys::SatGen::importCell(Yosys::RTLIL::Cell*, int)':
kernel/satgen.cc:1137:32: warning: 'undef_srst' may be used uninitialized in this function [-Wmaybe-uninitialized]
      std::tie(d, undef_d) = mux(srst, undef_srst, rval, undef_rval, d, undef_d);
                             ~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
kernel/satgen.cc:1123:32: warning: 'undef_en' may be used uninitialized in this function [-Wmaybe-uninitialized]
      std::tie(d, undef_d) = mux(en, undef_en, d, undef_d, old_q, undef_old_q);
                             ~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
kernel/satgen.cc:1109:32: warning: 'undef_srst' may be used uninitialized in this function [-Wmaybe-uninitialized]
      std::tie(d, undef_d) = mux(srst, undef_srst, rval, undef_rval, d, undef_d);
                             ~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[0m[  7%] Building libs/subcircuit/subcircuit.o
[  7%] Building libs/ezsat/ezsat.o
[  7%] Building libs/ezsat/ezminisat.o
[  8%] Building libs/minisat/Options.o
[  8%] Building libs/minisat/SimpSolver.o
[  9%] Building libs/minisat/Solver.o
[  9%] Building libs/minisat/System.o
[  9%] Building frontends/aiger/aigerparse.o
[ 10%] Building frontends/rpc/rpc_frontend.o
[ 10%] Building frontends/verific/verific.o
[ 10%] Building frontends/json/jsonparse.o
[ 11%] Building frontends/verilog/verilog_parser.tab.o
[... 2703 lines cut ...]
[ 94%] ABC: `` Compiling: /src/bdd/llb/llb1Cluster.c
[ 94%] ABC: `` Compiling: /src/bdd/llb/llb1Constr.c
[ 94%] ABC: `` Compiling: /src/bdd/llb/llb1Core.c
[ 94%] ABC: `` Compiling: /src/bdd/llb/llb1Group.c
[ 94%] ABC: `` Compiling: /src/bdd/llb/llb1Hint.c
[ 94%] ABC: `` Compiling: /src/bdd/llb/llb1Man.c
[ 94%] ABC: `` Compiling: /src/bdd/llb/llb1Matrix.c
[ 94%] ABC: `` Compiling: /src/bdd/llb/llb1Pivot.c
[ 94%] ABC: `` Compiling: /src/bdd/llb/llb1Reach.c
[ 94%] ABC: `` Compiling: /src/bdd/llb/llb1Sched.c
[ 94%] ABC: `` Compiling: /src/bdd/llb/llb2Bad.c
[ 94%] ABC: `` Compiling: /src/bdd/llb/llb2Core.c
[ 94%] ABC: `` Compiling: /src/bdd/llb/llb2Driver.c
[ 94%] ABC: `` Compiling: /src/bdd/llb/llb2Dump.c
[ 94%] ABC: `` Compiling: /src/bdd/llb/llb2Flow.c
[ 94%] ABC: `` Compiling: /src/bdd/llb/llb2Image.c
[ 94%] ABC: `` Compiling: /src/bdd/llb/llb3Image.c
[ 94%] ABC: `` Compiling: /src/bdd/llb/llb3Nonlin.c
[ 94%] ABC: `` Compiling: /src/bdd/llb/llb4Cex.c
[ 94%] ABC: `` Compiling: /src/bdd/llb/llb4Image.c
[ 94%] ABC: `` Compiling: /src/bdd/llb/llb4Nonlin.c
[ 94%] ABC: `` Compiling: /src/bdd/llb/llb4Sweep.c
[ 94%] ABC: `` Building binary: abc-bafd2a7
The command '/bin/sh -c make PREFIX=/install CONFIG=gcc ABCREV=bafd2a7 ABCURL=https://github.com/berkeley-abc/abc install -j$(nproc)' returned a non-zero code: 2
`

vijayank88, 2022-01-12:
@AndreKuru 
Have you ran `Install dependencies` from https://openroad.readthedocs.io/en/latest/main/README.html

vvbandeira, 2022-01-12:
@AndreKuru could you provide the output of `docker info`?
Please wrap the output here with triple backticks to get correct formatting.
Example:
```
output
```

AndreKuru, 2022-01-12:
`Docker info`:

```Client:
 Context:    default
 Debug Mode: false
 Plugins:
  buildx: Docker Buildx (Docker Inc., v0.7.1-docker)

Server:
 Containers: 15
  Running: 0
  Paused: 0
  Stopped: 15
 Images: 40
 Server Version: 20.10.12
 Storage Driver: overlay2
  Backing Filesystem: extfs
  Supports d_type: true
  Native Overlay Diff: false
  userxattr: false
 Logging Driver: json-file
 Cgroup Driver: systemd
 Cgroup Version: 2
 Plugins:
  Volume: local
  Network: bridge host ipvlan macvlan null overlay
  Log: awslogs fluentd gcplogs gelf journald json-file local logentries splunk syslog
 Swarm: inactive
 Runtimes: io.containerd.runc.v2 io.containerd.runtime.v1.linux runc
 Default Runtime: runc
 Init Binary: docker-init
 containerd version: 1407cab509ff0d96baa4f0eb6ff9980270e6e620.m
 runc version: v1.0.3-0-gf46b6ba2
 init version: de40ad0
 Security Options:
  seccomp
   Profile: default
  cgroupns
 Kernel Version: 5.15.13-arch1-1
 Operating System: Arch Linux
 OSType: linux
 Architecture: x86_64
 CPUs: 6
 Total Memory: 15.56GiB
 Name: kovir
 ID: 22EZ:V2BY:H5FM:UKUI:NX3H:MZGA:J4HE:7F3B:6FSO:PFS4:D4DC:VI36
 Docker Root Dir: /var/lib/docker
 Debug Mode: false
 Registry: https://index.docker.io/v1/
 Labels:
 Experimental: false
 Insecure Registries:
  127.0.0.0/8
 Live Restore Enabled: false
```

`git status`:
```
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
```

vvbandeira, 2022-01-18:
Thanks @AndreKuru.
Can you also try to clean the workspace before building?
```
git clean -d -x -i
git submodule foreach --recursive git clean -d -x -i
./build_openroad.sh
```

AndreKuru, 2022-01-19:
@vvbandeira, Thank you. I builded successfully after the clean.

closed 2022-01-19 by AndreKuru