The-OpenROAD-Project/OpenROAD-flow-scripts #308: ABC_AREA Synthesis strategies not optimizing the final area

vijayank88, 2022-01-12:
**Describe the bug**
* [Toolname] for runtime issues (Yosys ABC)
* [Design] for example design issues (ibex design)

**Expected behavior**
As per ABC_AREA synthesis strategies should generate better area optimization compare to ABC_SPEED. But for ibex design number of cells and area increased with ABC_AREA. Expecting better area optimization by flow.

**Environment (please complete the following information):**
Please only report issues for supported OSs.
 - OS: [e.g. CentOS 7]
 - OpenROAD-flow commit : v2.0-2318-gfed16d7ea 

**File Uploads**
In Makefile enable  DESIGN_CONFIG=./designs/sky130hd/ibex/config.mk.
Updated with ABC_AREA strategies and final ` Design area 194569 u^2 27% utilization.`
[ABC_AREA_design_area.txt](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/7853547/ABC_AREA_design_area.txt)
Synthesis statistics as follows:
[AREA_synth_stat.txt](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/7853574/AREA_synth_stat.txt)
Short view:
```
 Number of wires:              23916
   Number of wire bits:          24150
   Number of public wires:        2009
   Number of public wire bits:    2243
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:              23992
Chip area for module '\ibex_core': 146625.625600
```

Updated ABC_SPEED (Default) strategies and final `Design area 189641 u^2 28% utilization`
[ABC_SPEED_design_area.txt](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/7853550/ABC_SPEED_design_area.txt)
Synthesis Statistics as follows:
[SPEED_synth_stat.txt](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/7853571/SPEED_synth_stat.txt)
Short view:
```

   Number of wires:              15757
   Number of wire bits:          15991
   Number of public wires:        2009
   Number of public wire bits:    2243
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:              15833
Chip area for module '\ibex_core': 136121.801600
```

vijayank88, 2022-01-12:
With Ubuntu 20.04 and OpenROAD v2.0-2318-gfed16d7ea
For ABC_AREA and ABC_SPEED generating same results:
synthesis statistics:
```
 Number of wires:              15336
   Number of wire bits:          15570
   Number of public wires:        2009
   Number of public wire bits:    2243
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:              15412
Chip area for module '\ibex_core': 127095.644800
```
Design area:
./logs/sky130hd/ibex/base/2_1_floorplan.log:Design area 121998 u^2 19% utilization.
./logs/sky130hd/ibex/base/3_1_place_gp.log:Design area 132482 u^2 21% utilization.
./logs/sky130hd/ibex/base/3_3_resizer.log:Design area 132482 u^2 21% utilization.
./logs/sky130hd/ibex/base/3_3_resizer.log:Design area 154771 u^2 24% utilization.
./logs/sky130hd/ibex/base/3_4_opendp.log:Design area 154771 u^2 24% utilization.
./logs/sky130hd/ibex/base/4_1_cts.log:Design area 155158 u^2 25% utilization.
./logs/sky130hd/ibex/base/4_1_cts.log:Design area 155158 u^2 25% utilization.
./logs/sky130hd/ibex/base/4_1_cts.log:Design area 174760 u^2 28% utilization.
./logs/sky130hd/ibex/base/5_1_fastroute.log:Design area 174760 u^2 28% utilization.
./logs/sky130hd/ibex/base/6_report.log:Design area 174760 u^2 28% utilization.```

```
Is Operating system will affect Openroad flow design results? Or I have any setup issues at my end?

tspyrou, 2022-01-12:
Hi @vijayank88, 
For ABC_AREA vs ABC_SPEED, the measurement should be made at the end of the floorplan step.
To see the benefit of ABC_AREA to the end of the flow, you need to remove the .sdc or make it very loose.
Otherwise the repair design command will increase the area to meet timing regardless of the netlist produced earlier.

Can you measure the area in the floorplan step and close if the area looks good at that step?
If it does look good at that step you can try removing the sdc from the area based run to see a difference at the end of the flow.

If it doesn't look good assign back to me.

Tom

vijayank88, 2022-01-13:
@tspyrou 
Yes able to achieve better area if do relaxation in sdc file.
export ABC_AREA=1
Without .sdc file :
At floorplan: ```Design area 140231 u^2 19% utilization.```
End of the flow :```Design area 178306 u^2 24% utilization.```
2X value of clock_period :
At floorplan: ```Design area 140231 u^2 19% utilization.``` 
End of the flow: ```Design area 187482 u^2 26% utilization.```

export ABC_SPEED=1
With default sdc constraints: 
At floorplan: ```Design area 129385 u^2 19% utilization.```
End of the flow: ```Design area 189641 u^2 28% utilization```

dralabeing, 2022-01-26:
Looks like this is resolved.

closed 2022-01-26 by dralabeing