The-OpenROAD-Project/OpenROAD-flow-scripts #323: Add option for synthesis fan out

antonblanchard, 2022-01-24:
During synthesis we use `abc` to limit fan out. At the moment we just use the defaults, which is 10. This may not be ideal for some technologies, so it might make sense to have a `config.mk` variable to change the default via the abc script: `buffer -N [FAN_OUT]`. This is what Openlane does.

maliberty, 2022-01-24:
The direction is to move towards replacing ORFS with OL.  Do you have a need for this in ORFS or it more of a general note?  If the later I would be inclined to table this.

antonblanchard, 2022-01-25:
@maliberty thanks, I didn't realise that was the plan. Tabling it sounds fine.

There's a couple of reasons I like debugging in ORFS, such as being able to restart from an arbitrary place in the flow, eg `make clean_route; make` as well as it doing a reasonable job of dependency tracking for incremental rebuilds. I'll work to open a couple of issues against Openlane, unless it's already in plan.

maliberty, 2022-01-25:
OL does have -from/-to though it isn't as automatic as ORFS.  It will in the same session save to CURRENT_STEP where it left off in case of a failure.  @donn this is an area where OL might be improved.

maliberty, 2023-01-11:
The OL migration isn't really happening - do you still want this in ORFS?

vijayank88, 2023-07-17:
@maliberty I hope adding synthesis fanout will be an advantage.

maliberty, 2023-07-17:
I don't know what effect it will have.  I doesn't seem hard to add such - @vijayank88 do you want to do it?

rovinski, 2023-07-17:
Perhaps the most complete way of doing this is extracting it from the SDC. Maybe we could have an ORFS step which simply starts openroad, reads the .lib and .sdc, and then writes out the max fanout (either from SDC or library default).

I think the current hacky method of extracting clock period for ABC (e.g.)
https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/69a30ddc0e2da34486d177f89ecf752050dd8ad4/flow/platforms/asap7/config.mk#L60
gives rise to the problem seen in #1178. It might be more resilient to extract `ABC_PERIOD_IN_PS`, `ABC_LOAD_IN_FF` and others to a cached object file which is read by yosys.

maliberty, 2023-07-17:
@rovinski you can't read the SDC without a netlist and this is running before yosys has produced one.

rovinski, 2023-07-17:
Fair, I forgot that's why this was necessary.

vijayank88, 2023-07-18:
@maliberty what is the default Max fanout used during synthesis? 
Am planning to use `SYNTH_MAX_FANOUT` 
Or do you have other variable name to use, for document this?

maliberty, 2023-07-18:
I am not aware that there is one.  Presumably abc obeys the library max fanout.