The-OpenROAD-Project/OpenROAD-flow-scripts #372: Clarification of power unit during log execution

agostini01, 2022-03-02:
Hi,

First, thank you for your amazing work.

This is more of a question than an issue.

I would like to clarify what are the power units reported during the flow execution.

In the beginning of the flow, this is printed:

```
Default units for flow
 time 1ns
 capacitance 1fF
 resistance 1kohm
 voltage 1v
 current 1mA
 power 1nW
 distance 1um
```

Then, as the flow proceeds, we can see the following report cards.
```
==========================================================================
finish report_power
--------------------------------------------------------------------------
Group                  Internal  Switching    Leakage      Total
                          Power      Power      Power      Power
----------------------------------------------------------------
Sequential             1.76e-02   3.69e-03   1.38e-03   2.27e-02  39.4%
Combinational          1.26e-02   1.81e-02   4.06e-03   3.48e-02  60.6%
Macro                  0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
Pad                    0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
----------------------------------------------------------------
Total                  3.02e-02   2.18e-02   5.44e-03   5.75e-02 100.0%
                          52.6%      37.9%       9.5%

==========================================================================
finish report_design_area
--------------------------------------------------------------------------
Design area 243671 u^2 16% utilization.
```

Is this design estimated to consume  5.75e-02 nW?

What is the default switching activity assumed to estimate this value?

Is there a better place to verify summary results?

Perhaps printing the default units at the end of the flow execution would be helpful.

Thank you!

AnikBalo, 2022-03-07:
I am not sure, but maybe the power unit here is in Watt(W). I had run a few designs on Openroad and in commercial EDA tools. From those data, I am assuming  this power unit.

AnikBalo, 2022-03-23:
The unit is in Watt. I found this in OpenSTA documentation. 
[image]

jjcherry56, 2022-04-04:
I added "Watts" to the report so the units are obvious.

The only switchiing activity will be from clocked register outputs if the documented set_power_activity command is not used to say otherwise.

closed 2022-04-04 by jjcherry56