The-OpenROAD-Project/OpenROAD-flow-scripts #380: git clone not working in the docker container.

Unkraut3, 2022-03-08:
Let's start the flow scripts docker container: 

`docker run -it -u $(id -u ${USER}):$(id -g ${USER}) -v $(pwd)/flow/platforms:/OpenROAD-flow-scripts/flow/platforms:ro openroad/flow-scripts`

Now it would be nice to transfer data in and out from the docker container. So let's use git for that:

`git clone https://github.com/efabless/caravel_user_project.git`

An here is what we get:

`Cloning into 'caravel_user_project'...`
`remote: Enumerating objects: 2116, done.`
`remote: Total 2116 (delta 0), reused 0 (delta 0), pack-reused 2116`
`Receiving objects: 100% (2116/2116), 50.64 MiB | 4.59 MiB/s, done.`
`Resolving deltas: 100% (1099/1099), done.`
`fatal: unable to look up current user in the passwd file: no such user`
`Unexpected end of command stream`

Now google says this is a [bug in git](https://stackoverflow.com/questions/20271926/git-error-unable-to-look-up-current-user-in-the-passwd-file-no-such-user-wh). It has been fixed in 2015. Now we're in 2022. So what about using a less historic version of git in the docker? Or maybe a less historic operating system? I know you're also refusing to upgrade yosys to anything that is not outdated. Unlikely to happen.

vvbandeira, 2022-03-08:
@Unkraut3, you can use your local machine to clone the repository and mount additional docker volumes. This should allow you to continue using ORFS.

```
# first clone locally
git clone https://github.com/efabless/caravel_user_project.git
# then mount on docker
docker run -it -u $(id -u ${USER}):$(id -g ${USER}) -v $(pwd)/flow/platforms:/OpenROAD-flow-scripts/flow/platforms:ro -v <PATH_TO_CARAVEL>:/mnt/caravel_user_project openroad/flow-scripts
```

We use the latest git version available from the OS. We do have plans to update to newer OSes, but I cannot give you a timeline just yet.

As for Yosys, we are currently a couple of months away from the HEAD, if you want you can use the latest version at your own risk:
```
cd tools/yosys
git remote add upstream https://github.com/YosysHQ/yosys
git fetch upstream
git checkout upstream/master
```
Then use `build_openroad.sh --no_init [ARGS]` to avoid overwriting the change.

vvbandeira, 2022-03-10:
@Unkraut3 were you able to get the code in/out from the docker container with the above?

Unkraut3, 2022-03-11:
No, I'm trying to compile things locally. Some time ago after [ugrading outdated depencies](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/206) I managed to get the flow scripts to run. I again had the issue I reported [here](https://github.com/The-OpenROAD-Project/OpenROAD/issues/1186). This time on both Ubuntu and Debian systems. Also ABC does not compile out of the box, as the version used by OpenRoad is outdated. After changing the revision of ABC in build_openroad.sh compiling seems to work. You really seem to like seasoned software.

vvbandeira, 2022-03-14:
> No, I'm trying to compile things locally.

@Unkraut3 I think we are getting off-topic, the Issue title states a docker problem. I can understand your concerns with outdated OS/SW and we can try to address these concerns, please file additional issues for each problem.

Regarding the original issue, i.e., "[...] transfer data in and out from the docker container. [...]". Did you try my proposed solution of mounting additional folders into the docker container?
```
# first clone locally
git clone https://github.com/efabless/caravel_user_project.git
# then mount on docker
docker run -it -u $(id -u ${USER}):$(id -g ${USER}) -v $(pwd)/flow/platforms:/OpenROAD-flow-scripts/flow/platforms:ro -v <PATH_TO_CARAVEL>:/mnt/caravel_user_project openroad/flow-scripts
```

closed 2022-07-13 by vvbandeira