The-OpenROAD-Project/OpenROAD-flow-scripts #433: read_db not working for .odb file format with OpenROAD v2.0-3394-gd19162e82

vijayank88, 2022-04-19:
**Describe the bug**

* [Toolname] for runtime issues: openroad 
* Today I've updated ORFS repo. Earlier `read_db 6_final.odb` works fine and able to load it directly to GUI. But with latest `OpenROAD v2.0-3394-gd19162e82 ` its getting failed with below logs:
```
OpenROAD v2.0-3394-gd19162e82 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
openroad> read_db 4_1_cts.odb 
terminate called after throwing an instance of 'odb::ZException'
Stack trace:
 0# 0x000000000100A0D6 in /home/vijayan/OPENROAD_FLOW/ORFS_LOCAL/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 1# 0x00007F8B32D81400 in /lib64/libc.so.6
 2# gsignal in /lib64/libc.so.6
 3# abort in /lib64/libc.so.6
 4# __gnu_cxx::__verbose_terminate_handler() in /lib64/libstdc++.so.6
 5# 0x00007F8B3368FA06 in /lib64/libstdc++.so.6
 6# 0x00007F8B3368FA33 in /lib64/libstdc++.so.6
 7# 0x00007F8B3368FC53 in /lib64/libstdc++.so.6
 8# 0x0000000000F22B88 in /home/vijayan/OPENROAD_FLOW/ORFS_LOCAL/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 9# odb::dbDatabase::read(_IO_FILE*) in /home/vijayan/OPENROAD_FLOW/ORFS_LOCAL/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
10# ord::OpenRoad::readDb(char const*) in /home/vijayan/OPENROAD_FLOW/ORFS_LOCAL/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
11# 0x0000000001016C76 in /home/vijayan/OPENROAD_FLOW/ORFS_LOCAL/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
12# 0x00007F8B35D14EB2 in /lib64/libtcl8.5.so
13# 0x00007F8B35D5936C in /lib64/libtcl8.5.so
14# TclObjInterpProcCore in /lib64/libtcl8.5.so
15# 0x00007F8B35D14EB2 in /lib64/libtcl8.5.so
16# 0x00007F8B35D15F1E in /lib64/libtcl8.5.so
17# Tcl_EvalEx in /lib64/libtcl8.5.so
18# TclEvalObjEx in /lib64/libtcl8.5.so
19# 0x00007F8B35D1CA65 in /lib64/libtcl8.5.so
20# 0x00007F8B35D14EB2 in /lib64/libtcl8.5.so
21# 0x00007F8B35D5936C in /lib64/libtcl8.5.so
22# 0x00007F8B35D61647 in /lib64/libtcl8.5.so
23# TclEvalObjEx in /lib64/libtcl8.5.so
24# 0x00007F8B35D1C1D0 in /lib64/libtcl8.5.so
25# 0x00007F8B35D14EB2 in /lib64/libtcl8.5.so
26# 0x00007F8B35D5936C in /lib64/libtcl8.5.so
27# 0x00007F8B35D61647 in /lib64/libtcl8.5.so
28# TclEvalObjEx in /lib64/libtcl8.5.so
29# 0x00007F8B35D9C27F in /lib64/libtcl8.5.so
30# 0x00007F8B35D14EB2 in /lib64/libtcl8.5.so
31# 0x00007F8B35D5936C in /lib64/libtcl8.5.so
32# TclObjInterpProcCore in /lib64/libtcl8.5.so
33# 0x00007F8B35D14EB2 in /lib64/libtcl8.5.so
34# 0x00007F8B35D15F1E in /lib64/libtcl8.5.so
35# Tcl_EvalEx in /lib64/libtcl8.5.so
36# Tcl_Eval in /lib64/libtcl8.5.so
37# ord::tclAppInit(Tcl_Interp*) in /home/vijayan/OPENROAD_FLOW/ORFS_LOCAL/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
38# Tcl_Main in /lib64/libtcl8.5.so
39# main in /home/vijayan/OPENROAD_FLOW/ORFS_LOCAL/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
40# __libc_start_main in /lib64/libc.so.6
41# 0x0000000001006A97 in /home/vijayan/OPENROAD_FLOW/ORFS_LOCAL/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad

```

**Expected behavior**
read_db feature should work as previous OR version

**Environment (please complete the following information):**
Please only report issues for supported OSs.
 - OS: [e.g. CentOS 7]
 - OpenROAD-flow commit https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/commit/210c848914715563219a2ef9c87b0cd038a6d5cf

vijayank88, 2022-04-19:
I think its backward compatibility issue. After re-run the flow with latest version `make gui_final` works fine.
If its known issue mean please close this.
Thanks...

maliberty, 2022-04-19:
This was already fixed with https://github.com/The-OpenROAD-Project/OpenROAD/pull/1790

closed 2022-04-19 by maliberty