The-OpenROAD-Project/OpenROAD-flow-scripts #446: Makefile:655: *** KLayout not found in PATH

RajeshDattaSiu, 2022-04-30:
I am a new user and just installed the tool.  When I am running the 'make' command  to run an example  I am getting this error : 

``Makefile:655: *** KLayout not found in PATH.  Stop.``

It is not getting the klayout in the path.  I can see the following two lines in the makefile script:

KLAYOUT_CMD             ?= $(shell command -v klayout)
KLAYOUT_FOUND            = $(if $(KLAYOUT_CMD),,$(error KLayout not found in PATH))

So in my case KLAYOUT_CMD is not getting the correct path. I have already installed Klayout and put the whole folder in tools. But I am not sure how to link the  layout.

It will be a great help if I can get some suggestion to solve this.

vijayank88, 2022-04-30:
@RajeshDattaSiu 
Have you installed klayout?

RajeshDattaSiu, 2022-04-30:
> @RajeshDattaSiu Have you installed klayout?

Ya it was the klayout issue. I was able to solve it. Thanks..

closed 2022-04-30 by RajeshDattaSiu