The-OpenROAD-Project/OpenROAD-flow-scripts #463: Error during merge GDS using Klayout

GabbedT, 2022-05-24:
After running `make DESIGN_CONFIG=./designs/sky130hd/ibex/config.mk` everything is fine, until the GDS merge phase.
This is the error:
```
Elapsed time: 0:16.26[h:]min:sec. CPU time: user 16.11 sys 0.13 (99%). Peak memory: 369252KB.
cp results/sky130hd/ibex/base/5_route.sdc results/sky130hd/ibex/base/6_final.sdc
sed '/OR_DEFAULT/d' platforms/sky130hd/lef/sky130_fd_sc_hd.tlef > objects/sky130hd/ibex/base/klayout_tech.lef
sed 's,<lef-files>.*</lef-files>,<lef-files>/home/gabriele/OpenROAD-flow-scripts/flow/objects/sky130hd/ibex/base/klayout_tech.lef</lef-files> <lef-files>/home/gabriele/OpenROAD-flow-scripts/flow/platforms/sky130hd/lef/sky130_fd_sc_hd_merged.lef</lef-files>,g' platforms/sky130hd/sky130hd.lyt > objects/sky130hd/ibex/base/klayout.lyt
make: command: Command not found
Makefile:653: *** KLayout not found in PATH.  Arresto.
```

Makefile:
```
GDS_MERGED_FILE = $(RESULTS_DIR)/6_1_merged.$(STREAM_SYSTEM_EXT)
$(GDS_MERGED_FILE): $(RESULTS_DIR)/6_final.def $(OBJECTS_DIR)/klayout.lyt $(GDSOAS_FILES) $(WRAPPED_GDSOAS) $(SEAL_GDSOAS)
	$(call KLAYOUT_FOUND)
	($(TIME_CMD) stdbuf -o L $(KLAYOUT_CMD) -zz -rd design_name=$(DESIGN_NAME) \
	        -rd in_def=$< \
	        -rd in_files="$(GDSOAS_FILES) $(WRAPPED_GDSOAS)" \
	        -rd config_file=$(FILL_CONFIG) \
	        -rd seal_file="$(SEAL_GDSOAS)" \
	        -rd out_file=$@ \
	        -rd tech_file=$(OBJECTS_DIR)/klayout.lyt \
	        -rd layer_map=$(GDS_LAYER_MAP) \
	        -r $(UTILS_DIR)/def2stream.py) 2>&1 | tee $(LOG_DIR)/6_1_merge.log
``` 
Can you help me?

maliberty, 2022-05-25:
You need to install klayout ("KLayout not found in PATH").

GabbedT, 2022-05-25:
@vijayank88 I installed klayout a couple days ago, so the last version

maliberty, 2022-05-25:
you can echo $PATH right before it calls klayout to see if it is in the path or not.

GabbedT, 2022-05-25:
@maliberty Not sure if this is right:
```
echo $PATH
/home/gabriele/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
```

maliberty, 2022-05-25:
I meant in the Makefile (Makefile:653)

GabbedT, 2022-05-25:
@maliberty Sorry but I never worked on Makefile and i'm really new in the linux environment.
I did this: 
```
GDS_MERGED_FILE = $(RESULTS_DIR)/6_1_merged.$(STREAM_SYSTEM_EXT)
$(GDS_MERGED_FILE): $(RESULTS_DIR)/6_final.def $(OBJECTS_DIR)/klayout.lyt $(GDSOAS_FILES) $(WRAPPED_GDSOAS) $(SEAL_GDSOAS) 
        echo $PATH $(call KLAYOUT_FOUND)
	($(TIME_CMD) stdbuf -o L $(KLAYOUT_CMD) -zz -rd design_name=$(DESIGN_NAME) \
	        -rd in_def=$< \
	        -rd in_files="$(GDSOAS_FILES) $(WRAPPED_GDSOAS)" \
	        -rd config_file=$(FILL_CONFIG) \
	        -rd seal_file="$(SEAL_GDSOAS)" \
	        -rd out_file=$@ \
	        -rd tech_file=$(OBJECTS_DIR)/klayout.lyt \
	        -rd layer_map=$(GDS_LAYER_MAP) \
	        -r $(UTILS_DIR)/def2stream.py) 2>&1 | tee $(LOG_DIR)/6_1_merge.log
```
And the output is still:
```
[INFO][FLOW] Using platform directory ./platforms/sky130hd
make: command: Command not found
make: command: Command not found
make: command: Command not found
make: command: Command not found
Makefile:654: *** KLayout not found in PATH.  Arresto.
```
I'm pretty sure that i did something wrong. l

maliberty, 2022-05-25:
Put it on its own line so it executes first:
```
 echo $PATH
 $(call KLAYOUT_FOUND)
 ($(TIME_CMD) stdbuf -o L $(KLAYOUT_CMD) -zz -rd design_name=$(DESIGN_NAME) \
```

maliberty, 2022-05-25:
Be sure to use tabs and not spaces at the line start - make is finicky about it.

GabbedT, 2022-05-25:
@maliberty Thanks, now the output is:
```
[INFO][FLOW] Using platform directory ./platforms/sky130hd
make: command: Command not found
make: command: Command not found
make: command: Command not found
make: command: Command not found
Makefile:656: the «(/usr/bin/time» of the target doesn't  does not match the model of the target itself
Makefile:656: the «-f» of the target doesn't  does not match the model of the target itself
Makefile:656: the «'Elapsed» of the target doesn't  does not match the model of the target itself
Makefile:656: the «time» of the target doesn't  does not match the model of the target itself
make: command: Command not found
Makefile:655: *** KLayout not found in PATH.  Arresto.
```

maliberty, 2022-05-25:
This is hard to do remotely.  With the original makefile would you try 'make versions.txt' and then post the contents of the generated file.

maliberty, 2022-05-25:
Also at your unix prompt do "command -v klayout" and post the results

GabbedT, 2022-05-25:
@maliberty I adjusted the Makefile, everything is highlighted so it should be correct, the output however is still:
```
[INFO][FLOW] Using platform directory ./platforms/sky130hd
make: command: Command not found
make: command: Command not found
make: command: Command not found
make: command: Command not found
Makefile:655: *** KLayout not found in PATH.  Arresto.
```

GabbedT, 2022-05-25:
Prompt: command -v klayout
Output: /usr/bin/klayout

GabbedT, 2022-05-25:
Prompt: make -version.txt
Output: 
```
GNU Make 4.2.1
Compiled for x86_64-pc-linux-gnu
Copyright (C) 1988-2016 Free Software Foundation, Inc.
 GPLv3 + license: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>.
 This program is free software: you are free to modify and redistribute it.
 There is NO WARRANTY, to the extent permitted by current regulations.
 ```

maliberty, 2022-05-25:
not "make -version.txt" but "make version.txt"

GabbedT, 2022-05-25:
> not "make -version.txt" but "make version.txt"

If you mean into the prompt this is the output:
```
gabriele@Gabbed:~/OpenROAD-flow-scripts/flow$ make version.txt
[INFO][FLOW] Using platform directory ./platforms/nangate45
make: command: Command not found
make: command: Command not found
make: command: Command not found
make: ***  make no rule to make target «version.txt».  Arresto.
```

maliberty, 2022-05-25:
Sorry "make version**s**.txt"

GabbedT, 2022-05-25:
@maliberty 
```
gabriele@Gabbed:~/OpenROAD-flow-scripts/flow$ make versions.txt
[INFO][FLOW] Using platform directory ./platforms/nangate45
make: command: Command not found
make: command: Command not found
make: command: Command not found
make: command: Command not found
/bin/sh: 1: zz: not found
make: [Makefile:302: versions.txt] Error 127 (ignored)
```

maliberty, 2022-05-26:
What OS are you using?  What shell?

You might be better off using a docker container because something seems very wrong with your environment and I can't really understand it remotely.

GabbedT, 2022-05-26:
> What OS are you using? What shell?
> 
> You might be better off using a docker container because something seems very wrong with your environment and I can't really understand it remotely.

OS: Ubuntu 20.04
Shell:
```
gabriele@Gabbed:~$ echo $SHELL
/bin/bash
```
```
gabriele@Gabbed:~$ echo $0
bash
```
```
gabriele@Gabbed:~$ $SHELL --version
GNU bash, versione 5.0.17(1)-release (x86_64-pc-linux-gnu)
```
Should I build the OpenROAD flow with docker? So this guide: https://openroad.readthedocs.io/en/latest/user/BuildWithDocker.html

vijayank88, 2022-05-26:
@GabbedT 
Have you installed required development packages? https://github.com/The-OpenROAD-Project/OpenROAD#install-dependencies

GabbedT, 2022-05-26:
@vijayank88 Yes, i did everything step by step

maliberty, 2022-05-26:
I think it is worth a try in docker to see if you have similar issues or not.

GabbedT, 2022-05-27:
@maliberty Yes! The test flow now works, thanks for helping me!

closed 2022-05-27 by GabbedT