The-OpenROAD-Project/OpenROAD-flow-scripts #467: No rule to make target `gui_6_final.odb'

GabbedT, 2022-05-27:
Hi, first of all this is my second issue report in two days, so i'm sorry if i'm opening too many.
I'm currently learning the OpenROAD flow and following this guide: https://openroad.readthedocs.io/en/latest/tutorials/FlowTutorial.html
I have all the dependencies installed, the environment set up and I'm running OpenROAD with Docker.

The design synthesis after running this command: `make DESIGN_CONFIG=./designs/sky130hd/ibex/config.mk` was successful.
However, when i try to view results and logs with this command: `make gui_final` (in flow directory)
```
bash-4.2$ make gui_final
[INFO][FLOW] Using platform directory ./platforms/nangate45
make: *** No rule to make target `gui_6_final.odb', needed by `gui_final'.  Stop.
```
It's also strange that it's using the **nandgate45** when i synthetised ibex with **sky130hd**,
I also uncommented the DESIGN_CONFIG like that:
```
# DESIGN_CONFIG=./designs/sky130hd/gcd/config.mk
DESIGN_CONFIG=./designs/sky130hd/ibex/config.mk
# DESIGN_CONFIG=./designs/sky130hd/aes/config.mk
```
I noticed that there's also this uncommented
```
# Default design
DESIGN_CONFIG ?= ./designs/nangate45/gcd/config.mk
```
I tried to comment that but the same output shows up.

Thanks in advance for helping me another time.

maliberty, 2022-05-27:
try: make DESIGN_CONFIG=./designs/sky130hd/ibex/config.mk gui_final

GabbedT, 2022-05-28:
@maliberty 
Now the pdk changes but the error is still the same.
```
bash-4.2$ make DESIGN_CONFIG=./designs/sky130hd/ibex/config.mk gui_final
[INFO][FLOW] Using platform directory ./platforms/sky130hd
make: *** No rule to make target `gui_6_final.odb', needed by `gui_final'.  Stop
```

maliberty, 2022-05-31:
Do you actually have a final odb file to open?

GabbedT, 2022-05-31:
@maliberty Doesn't the synthesis generate automatically that file? If not how can I do it?

vijayank88, 2022-06-02:
@GabbedT 
Other way just remove `#` from `Makefile` for ibex.
Do `make clean_all` and run again with `make`
Then try `make gui_final`

GabbedT, 2022-06-02:
@vijayank88 When i run `make clean_all` i get this:
```
~/OpenROAD-flow-scripts/flow$ make clean_all
[INFO][FLOW] Using platform directory ./platforms/sky130hd
/bin/sh: 0: Illegal option -o pipefail
/bin/sh: 0: Illegal option -o pipefail
/bin/sh: 0: Illegal option -o pipefail
/bin/sh: 0: Illegal option -o pipefail
/bin/sh: 0: Illegal option -o pipefail
/bin/sh: 0: Illegal option -o pipefail
/bin/sh: 0: Illegal option -o pipefail
/bin/sh: 0: Illegal option -o pipefail
rm -f  ./results/sky130hd/ibex/base/1_*.v ./results/sky130hd/ibex/base/1_synth.sdc
/bin/sh: 0: Illegal option -o pipefail
/bin/sh: 0: Illegal option -o pipefail
make: *** [Makefile:382: clean_synth] Errore 2
```
I also already left the `DESIGN_CONFIG` for ibex.

vvbandeira, 2022-06-02:
@GabbedT see https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/discussions/330 for a solution

GabbedT, 2022-06-02:
@vvbandeira My bad i didn't started the docker.

GabbedT, 2022-06-03:
@vijayank88  Now the output is:
```
[INFO][FLOW] Using platform directory ./platforms/sky130hd
ODB_FILE=./results/sky130hd/ibex/base/6_final.odb /OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad -gui  ./scripts/gui.tcl
This plugin does not support propagateSizeHints()
OpenROAD GITDIR-NOTFOUND 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
Loading spef
```

vijayank88, 2022-06-04:
@GabbedT 
Have you updated packages as per https://openroad.readthedocs.io/en/latest/user/BuildLocally.html#prerequisites?

GabbedT, 2022-06-04:
@vijayank88 Yes, everything is updated to the most recent version.

vijayank88, 2022-06-04:
`git --version` ?

GabbedT, 2022-06-04:
@vijayank88 2.25.1

vvbandeira, 2022-06-04:
> @vijayank88 Now the output is:
>
> ```
> [INFO][FLOW] Using platform directory ./platforms/sky130hd
> ODB_FILE=./results/sky130hd/ibex/base/6_final.odb [...]

These messages look ok if you are running inside docker. Do you have any error messages or problems completing the flow?

vvbandeira, 2022-06-05:
@GabbedT so can we close the issue or did I miss something?

GabbedT, 2022-06-06:
@vvbandeira The problem is that i can't visualize the area/timing/power reports with the commands `report_design_area`, `report_worst_slack` etc... 

Basically, after the completion of the flow, I run the command: `make DESIGN_CONFIG=./designs/sky130hd/ibex/config.mk gui_final` and this output appears:

```
[INFO][FLOW] Using platform directory ./platforms/sky130hd
ODB_FILE=./results/sky130hd/ibex/base/6_final.odb /OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad -gui  ./scripts/gui.tcl
This plugin does not support propagateSizeHints()
OpenROAD GITDIR-NOTFOUND 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
Loading spef
```

After `loading spef` basically nothing happens when I try to run the reports commands, I'm not even sure that this is the correct procedure.

If I'm correct in the terminal, the result data should appear, right?

vijayank88, 2022-06-06:
> @GabbedT Other way just remove `#` from `Makefile` for ibex and keep `#` behind other `DESIGN_CONFIG` Do `make clean_all` and run again with `make` Then try `make gui_final`

Have you tried above option?

Is that local or using inside docker?

GabbedT, 2022-06-06:
@vijayank88 Yes, I already tried that. I'm using docker

vvbandeira, 2022-06-06:
@GabbedT how are you starting docker? You will probably need something like:
```
xhost +local:docker
docker \
   run \
   --tty \
   --interactive \
   --user $(id -u ${USER}):$(id -g ${USER}) \
   --net=host \
   --ipc=host \
   --env DISPLAY=${DISPLAY} \
   --volume $(pwd)/flow/platforms:/OpenROAD-flow-scripts/flow/platforms:ro \
   --volume $(pwd)/flow/scripts:/OpenROAD-flow-scripts/flow/scripts:ro \
   --volume ${HOME}:/home/${USER} \
   --volume /tmp/.X11-unix:/tmp/.X11-unix \
   --volume ${HOME}/.Xauthority:/.Xauthority \
   --security-opt seccomp=unconfined \
   --privileged \
   openroad/flow-scripts \
   bash
xhost -local:docker
```

GabbedT, 2022-06-06:
@vvbandeira Yes! Now the gui opens with the synthetized design, it works fine except the `TCL command` section: I can't really input any command from the keyboard. Should I close the issue and open another related to that problem?

vvbandeira, 2022-06-06:
You can probably just close this one. The keyboard issue is known already, you will need to install some extra packages to your docker image. See https://github.com/The-OpenROAD-Project/OpenROAD/issues/1834#issuecomment-1135123666
I have this on my list to update the docs but did not have a chance yet.

GabbedT, 2022-06-06:
@vvbandeira Perfect, thank you again

closed 2022-06-06 by GabbedT