The-OpenROAD-Project/OpenROAD-flow-scripts #469: Cannot run floorpan for pulpino design

minghungumich, 2022-05-29:
**Describe the bug**
I build the tool locally and can run the tutorial design successfully.
Then when I try to import my pulpino design, it stuck at macro placement if manual macro cfg file is not given.
But it should run the automatic macro placement by default.
* [Toolname] stuck at macro placement (`mpl`)
* [Flow] 2_4_floorplan_macro
* [Design] pulpino RISC-V system with sky130 sram

**Expected behavior**
Floorplan complete and run the flow to the end

**Screenshots**
```
(/usr/bin/time -f 'Elapsed time: %E[h:]min:sec. CPU time: user %U sys %S (%P). Peak memory: %MKB.' /home/kevinchen/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad -exit -no_init  ./scripts/macro_place.tcl -metrics ./logs/sky130hd/pulpino/base/2_4_mplace.json) 2>&1 | tee ./logs/sky130hd/pulpino/base/2_4_mplace.log
OpenROAD v2.0-3729-g713ef39fc
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO MPL-0005] Found 16 macros.
[INFO MPL-0009] West pins 114.
[INFO MPL-0009] East pins 113.
[INFO MPL-0009] North pins 168.
[INFO MPL-0009] South pins 168.
^Cmake: *** [Makefile:441: results/sky130hd/pulpino/base/2_4_floorplan_macro.odb] Error 130
```

**Environment (please complete the following information):**
Please only report issues for supported OSs.
 - OS: CentOS 7
 - OpenROAD-flow commit: `1ba989a`

**File Uploads**

**Additional context**

minghungumich, 2022-05-29:
@mithro @msaligane I am facing floorpan issue when macro cfg is not provided.

maliberty, 2022-05-29:
please use make <step>_issue to package your test case.

vijayank88, 2022-05-30:
@minghungumich 
Can you please update your repo and try. 
With latest version I am not able to reproduce issue.

```
./run-me-pulpino-sky130hd-base.sh 
OpenROAD v2.0-3824-gee9b117da 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: ./platforms/sky130hd/lef/sky130_fd_sc_hd.tlef
[INFO ODB-0223]     Created 13 technology layers
[INFO ODB-0224]     Created 25 technology vias
[INFO ODB-0226] Finished LEF file:  ./platforms/sky130hd/lef/sky130_fd_sc_hd.tlef
[INFO ODB-0222] Reading LEF file: ./platforms/sky130hd/lef/sky130_fd_sc_hd_merged.lef
[INFO ODB-0225]     Created 441 library cells
[INFO ODB-0226] Finished LEF file:  ./platforms/sky130hd/lef/sky130_fd_sc_hd_merged.lef
[INFO ODB-0222] Reading LEF file: ./designs/sky130hd/pulpino/lef/sky130_sram_2kbyte_1rw_32x512_8.lef
[INFO ODB-0225]     Created 1 library cells
[INFO ODB-0226] Finished LEF file:  ./designs/sky130hd/pulpino/lef/sky130_sram_2kbyte_1rw_32x512_8.lef
number instances in verilog is 84692
[WARNING IFP-0028] Core area lower left (2.000, 2.000) snapped to (2.300, 2.720).
[INFO IFP-0001] Added 1492 rows of 8823 sites.
[INFO RSZ-0026] Removed 520 buffers.
```

minghungumich, 2022-05-30:
@vijayank88 The testcases is updated and I have confirmed I can reproduce the same log by running the provided shell script. [Macro Place Testcase](https://github.com/minghungumich/Testcase/blob/main/macro_place_pulpino_sky130hd_base_2022-05-30_14-58.tar.gz)

vijayank88, 2022-06-01:
@maliberty 
Issue confirmed. Can you please have a look into that or please assign to right people?

minghungumich, 2022-06-02:
@ravi-varadarajan This one should be reproducible. [Macro Place Testcase](https://github.com/minghungumich/Testcase/blob/main/macro_place_pulpino_sky130hd_base_2022-05-30_14-58.tar.gz)

vijayank88, 2022-06-17:
@ravi-varadarajan 
are you able to reproduce the issue with above `Macro Place Testcase`?

ravi-varadarajan, 2022-06-17:
Apologies for the delay. I will investigate and get back on this

vijayank88, 2023-06-13:
@minghungumich 
Is this issue still valid?
We're not finding the test case.

minghungumich, 2023-06-13:
This is obsolete. I will close the issue.

closed 2023-06-13 by minghungumich