The-OpenROAD-Project/OpenROAD-flow-scripts #472: Boost Serialization Version conflict

Sinbiosis, 2022-06-02:
**Describe the bug**
Running `./build_openroad.sh --local` results in issue with configuration of boost_serialization. Instead I get the following

```
CMake Error at /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake:117 (find_package):
  Could not find a configuration file for package "boost_serialization" that
  exactly matches requested version "1.71.0".

  The following configuration files were considered but not accepted:

/usr/local/lib/cmake/boost_serialization-1.76.0/boost_serialization-config.cmake, version: 1.76.0

Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake:182 (boost_find_component)
  /usr/share/cmake-3.16/Modules/FindBoost.cmake:443 (find_package)
  src/drt/CMakeLists.txt:47 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/element/src/BMI-repos/OpenROAD-flow-scripts/tools/OpenROAD/build/CMakeFiles/CMakeOutput.log".
See also "/home/element/src/BMI-repos/OpenROAD-flow-scripts/tools/OpenROAD/build/CMakeFiles/CMakeError.log".
```

**Expected behavior**
Running above compiles openroad locally.

**Screenshots**
N/A

**Environment (please complete the following information):**
Please only report issues for supported OSs.
 - OS: Ubuntu 20.04.4 LTS
 - OpenROAD-flow commit a9135b34ffdaa0c04b18a4423d038cd98cc8827f

**File Uploads**
N/A

**Additional context**
N/A

Sinbiosis, 2022-06-02:
Running the following from [here](https://github.com/phylo42/rappas2/issues/4) worked for me:

`sudo apt install build-essential cmake libboost-dev libboost-serialization-dev libboost-filesystem-dev libboost-iostreams-dev libboost-program-options-dev zlib1g-dev rapidjson-dev libquadmath0`

vvbandeira, 2022-06-02:
@Sinbiosis did you follow the installation guide? We build and test with Ubuntu 20.04 using docker, so I suspect it is something with your environment.
https://openroad.readthedocs.io/en/latest/user/BuildLocally.html

vvbandeira, 2022-10-03:
Closing as stale. Please re-open or file a new issue in case this is still relevant.

closed 2022-10-03 by vvbandeira

dansharkey, 2024-01-16:
I'm having the same (similar issue) on a brand new Ubuntu 23.10 (mantic) install. Following the additional apt installs from above (Sinbiosis comment from 2nd Jun 2022) failed to resolve the issue.

Error message is almost identical to above, but instead of boost_serialization v1.76.0, I have version 1.80.0:

```
CMake Error at /home/dan/Work/vlsi/tools/OpenROAD-flow-scripts/dependencies/lib/cmake/Boost-1.80.0/BoostConfig.cmake:141 (find_package):
  Found package configuration file:

    /home/dan/Work/vlsi/tools/OpenROAD-flow-scripts/dependencies/lib/cmake/boost_serialization-1.80.0/boost_serialization-config.cmake

  but it set boost_serialization_FOUND to FALSE so package
  "boost_serialization" is considered to be NOT FOUND.  Reason given by
  package:

  No suitable build variant has been found.

  The following variants have been tried and rejected:

  * libboost_serialization.so.1.80.0 (shared, Boost_USE_STATIC_LIBS=ON)
```

I have run the `setup.sh` `etc/DependencyInstaller.sh` and `build_openroad.sh` scripts.

vvbandeira, 2024-01-16:
@dansharkey 
Could you file a new GH issue and follow all the instructions to provide a complete picture of your env?