The-OpenROAD-Project/OpenROAD-flow-scripts #473: [Build] Issues building OpenROAD Locally

Sinbiosis, 2022-06-02:
**Describe the bug**
Running `./build_openroad.sh --local` results in issue during build. It is difficult to diagnose issue from logs. I'm seeing errors about  no matching function call:

```
/home/element/src/BMI-repos/OpenROAD-flow-scripts/tools/OpenROAD/build/src/dbSta/src/CMakeFiles/dbSta.dir/dbStaTCL_wrap.cxx: In function ‘void set_timing_derate_cell_cmd(const sta::LibertyCell*, sta::TimingDerateType, sta::PathClkOrData, const sta::RiseFallBoth*, const EarlyLate*, float)’:
/home/element/src/BMI-repos/OpenROAD-flow-scripts/tools/OpenROAD/build/src/dbSta/src/CMakeFiles/dbSta.dir/dbStaTCL_wrap.cxx:4727:75: error: no matching function for call to ‘sta::Sta::setTimingDerate(const sta::LibertyCell*&, sta::TimingDerateType&, sta::PathClkOrData&, const sta::RiseFallBoth*&, const EarlyLate*&, float&)’
 4727 |   Sta::sta()->setTimingDerate(cell, type, clk_data, rf, early_late, derate);
```

This issue popped up after fixing the issue I listed [here (Boost Serialization Version conflict)](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/472)

**Expected behavior**
Build completes for OpenROAD Local.

**Screenshots**
N/A

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.4 LTS
 - OpenROAD-flow commit a9135b34ffdaa0c04b18a4423d038cd98cc8827f

**File Uploads**
[CMakeError.log](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/8827619/CMakeError.log)
[CMakeOutput.log](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/8827620/CMakeOutput.log)

**Additional context**
N/A

vvbandeira, 2022-10-07:
@Sinbiosis, apologies for the delay. Do you still see this issue with the latest code?

Sinbiosis, 2022-10-08:
I will check once I get back to my main computer.

vvbandeira, 2022-11-27:
Closing due to lack of activity. Please open a new issue in case the problem is still relevant and occurring with the latest commit.

closed 2022-11-27 by vvbandeira