The-OpenROAD-Project/OpenROAD-flow-scripts #477: PDN breaks after defining global connect for macro pg pins

minghungumich, 2022-06-05:
**Describe the bug**
I build the tool locally and can run the tutorial design successfully.
Then when I try to import my ibex implementation, it stuck at macro placement if manual macro cfg file is not given.
But it should run the automatic macro placement by default.
* [Toolname] wrong functionality at pdngen (`pdn`)
* [Flow] 2_6_floorplan_pdn
* [Design] ibex RISC-V system with sky130 sram

**Expected behavior**
Power stripe and straps should not be destroyed. 

**Screenshots**
When using default `pdn.tcl`
[image]
When using self defined `pdn.tcl`
(Only add the following api command before global_connect)
```
add_global_connection -defer_connection -net {VDD} -inst_pattern {.*} -pin_pattern {^vccd1$}
add_global_connection -defer_connection -net {VSS} -inst_pattern {.*} -pin_pattern {^vssd1$}
```
[image]

**Environment (please complete the following information):**

Please only report issues for supported OSs.

 - OS: CentOS 7
 - OpenROAD-flow commit: `1ba989a`

**File Uploads**
[PDN Testcase](https://github.com/minghungumich/Testcase/blob/main/pdn_pulpino_sky130hd_base_2022-06-05_16-30.tar.gz)

**Additional context**

minghungumich, 2022-06-05:
@mithro @msaligane I am facing pdn issue when manual pdn cfg is defined.

arlpetergadfort, 2022-06-05:
@minghungumich is the issue that there are no straps on met4 over the macro? That would be because PDNGEN will not add straps on layers that the macro has metal in (since the highest pins are in met4, the only metal that can connect is on met5).

minghungumich, 2022-06-06:
@arlpetergadfort I know there are blockages in macro.lef so PDNGEN will not add the strap over it. I think the main issue is after I add extra global connect for macro pg pins, the connection should be a little difference with first figure (default setting). There are only vias and no straps in the second figure (manual setting). Maybe it is my klayout displaying error, but I did find there are lots of difference after adding global connect in DEF file. I use the default sky130hd pdn.tcl setting as shown below.

arlpetergadfort, 2022-06-06:
> @arlpetergadfort I know there are blockages in macro.lef so PDNGEN will not add the strap over it. I think the main issue is after I add extra global connect for macro pg pins, the connection should [...]

When I ran it, the straps were present after the pdn step (with the run script in the issue you uploaded). You may want to check the layout in OpenROAD to see when the straps disappear.

Global connect doesn't add or remove anything physical it just tells the database what net to connect to what pins.

minghungumich, 2022-06-06:
@arlpetergadfort Could you share your screenshot with straps present? If that, this might be my klayout problem.

arlpetergadfort, 2022-06-06:
Here is the screenshot of the upper left corner.
[image]

You can see the straps on met4 and met5. There are vias connected the memory to the nets on met5 (not visible in this image).

minghungumich, 2022-06-06:
Thanks! It looks like my klayout displaying problem. I can see those straps in openroad gui.

closed 2022-06-06 by minghungumich